nasa/XPlaneConnect #51: Determine Simulink Architecture

teubert, 2015-04-09:


nlsn, 2016-03-02:
Any thoughts about a Simulink architecture?

I saw that there are a couple different implementations available on Mathworks File Exchange, but I haven't tried using them.

jason-watkins, 2016-03-02:
Hey Nelson,

To be honest, I rather doubt a Simulink interface for XPC is going to happen in the near future. I worked on the project as a summer intern, so I'm not officially on the project anymore at all, although I do try to keep contributing when I can. And last I heard, Chris (@teubert) didn't have any time allocated for XPC development either, so there's not really anyone to do the work.

That said, it was my hope that it would be relatively easy to use XPC within Simulink by simply embedding MATLAB functions, but I never learned enough Simulink to know if that was realistic. If you have a particular use case in mind, I would be interested to hear your thoughts on what we should do to better support simulink.

Best,
Jason

P.S., Based on the F-15 avatar, are you the same Nelson Brown that works at Armstrong?

nlsn, 2016-03-02:
I have a vague notion of finding a student for the fall who wants to put together a wake-surfing simulation of two airplanes that can be published, replicated, and modified.

We have a working sim at Armstrong, and we are planning research flights this summer. So we will hopefully have some validation datasets soon.

I don't think X-plane makes any attempt to model wakes, and I'm not at all sure that a wake model could be grafted on with XPC.

jason-watkins, 2016-03-02:
Unfortunately, the internals of X-Plane would make that scenario pretty tough to implement using XPC. We can get and set just about any variable in the X-Plane sim, but we there isn't an easy way to add in new simulation parameters. You could probably figure out how to get something working, but I can think of a few issues that are likely to crop up, and in the end it might be more frustrating than helpful.

It sounds like an interesting problem to tackle though. I wouldn't mind taking a more through look at what it would take to add your model into X-Plane, but I would need to see the details of the wake simulation to really get an idea of what's involved. I'm guessing the existing sim isn't cleared for public release, in which case there's not much I can do. If it happens that you can send me some more details about the wake sim, I'd be happy to discuss supporting your use case further.

In any case, I think that grafting a wake model into X-Plane is likely to be a harder problem than interacting with XPC in Simulink.

nlsn, 2016-03-02:
Thanks for the insight. The wake model is a Simulink implementation of a
published wake model. The hard part is Armstrong software release policy.
We have a modified TCM sim version. Maybe TCM could be connected to X-Plane
for cockpit interface and vis, but keep the EOM in Simulink.
TCM: https://software.nasa.gov/software/LAR-18322-1

teubert, 2016-03-02:
It sounds like Jason's already answered most of your questions. I would just add a few things:

  *   Using X-Plane for your visuals (Out-the window display, multiple aircraft scenario visualization, control panel) is a definite possibility. That would not be very difficult to implement.
  *   I believe there is already a wake turbulence model implemented in X-Plane. I'm not sure what model they use, but I could look into that if you would like me too.  Here's a video of X-Plane Wake turbulence in action: https://www.youtube.com/watch?v=pgaAQAWwthA
  *   I agree with Jason's assessment that it might be difficult to implement your wake turbulence model with the X-Plane flight simulator.
  *   Jason is right- Unfortunately, right now there is no funding for XPC support. That said Jason will be likely returning this summer as an intern again. At that point he might be able to provide some additional support.

Hope this helps,
Chris

nlsn, 2016-03-02:
Well now I'm curious what wake vortex models are implemented in X-Plane. But please don't trouble yourself with investigating just to answer my curiosity. Since the X-Plane code is closed, I think it would be an uphill battle to convince people the model is good enough to rely on for research.