nasa/XPlaneConnect #73: Extra line from waypoint

teubert, 2015-05-14:
When running the MATLAB test scripts on mac, an extra red line is coming from the waypoint immediately in front of the vehicle (see screen shot). This line goes into the ground at a steep angle.

[image]

jason-watkins, 2015-05-14:
Can't reproduce on Windows. I suspect that the issue lies in a difference in the Mac and Windows OpenGL implementations.

teubert, 2015-05-14:
I cannot replicate the three line problem, but after further investigation I have found that 1. The second line only occurs once the test scripts have been run twice. and 2) The line ends at waypoint1 (see attached image)

[image]

jason-watkins, 2015-05-15:
So far, we haven't been able to reproduce this a second time.

closed 2015-05-15 by jason-watkins