nasa/XPlaneConnect #76: I got this error when I used matlab :Error using XPlaneConnect.getDREFs (line 27) [getDREFs] ERROR: Multiple clients open. You must specify which client to use.

magedmikhail, 2015-06-16:


jason-watkins, 2015-06-16:
Hi magedmikhail.

This is usually caused by calling `openUDP` more than once, and not specifying which socket you want to use in subsequent calls.

Could you post an [SSCCE](http://sscce.org/)?

magedmikhail, 2015-06-16:
Dear all
please Can you tell me the  steps to set the connection between matalb and
x plane.

1. I have  - x Plane 10 ready

2.  I Copied the X-Plane Plugin folder (xpcPlugin/XPlaneConnect) to the
plugin directory ([X-Plane Directory]/Resources/plugins/)

3. matlab ready for Example1

what I should to do for x plane setting option for input output and net
connection(IP  for data output and UDP) how you set them up  to match with
matlab

Socket = openUDP( 49005); is this correct or Socket = openUDP(
'127.0.0.1',49005);

getDREFs('sim/test/test_float')  (My problem here)

-- 

Thanks,
Dr. Maged mikhail
Maged B. Mikhail  Ph.D.
Tennessee State University
Dept. of Electrical and Computer Engr.
Intelligent Control Systems Lab Rm. 201
Work: (615) 963-5328
Cell: (615) 710-6745
E_mail: mmikhail@my.tnstate.edu <mmikhail@mytsu.tnstate.edu>

jason-watkins, 2015-06-16:
`Socket = openUDP( 49005);` does appear to be a bug in the sample script. I have corrected it for the next release. `Socket = openUDP( '127.0.0.1',49005);` is indeed correct.

Could you verify that you get the same error with a freshly opened copy of MATLAB? Because the example script does not include robust error handling, it's possible for sockets to be abandoned but left open if some other error causes the example script to exit early.

teubert, 2015-06-22:
Hello Maged Mikhail,

Were you able to get this working? I wanted to check in and make sure you got all your questions answered.

~Chris

dangmoran1991, 2015-07-06:
Got the same problem here, and can not solve....
I also have:
1.  x Plane 10 ready
2.  I Copied the X-Plane Plugin folder (xpcPlugin/XPlaneConnect) to the
plugin directory ([X-Plane Directory]/Resources/plugins/)
3. matlab ready for Example1

and the problem is like:
[image]

Thank you.
Moran

jason-watkins, 2015-07-06:
Hi Moran.

I've just pushed an update to the repository that fixes several issues with the MATLAB example script. Could you update to the latest version of master and let me know if it works for you?

Thanks,
Jason

jason-watkins, 2015-07-08:
@ magedmikhail @dangmoran1991 

Since I haven't heard anything more, I'm going to hope that no news is good news and you've both gotten things up and running. Feel free to comment here or open a new issue if we can provide any additional help.

Thanks,
Jason

closed 2015-07-08 by jason-watkins