nasa/XPlaneConnect #77: Error when using "interpreted MATLAB Fcn" block in Simulink

jihoon7942, 2015-07-06:
Java exception occurred: 
java.io.IOException: No response received.
	at gov.nasa.xpc.XPlaneConnect.getDREFs(XPlaneConnect.java:299)

jason-watkins, 2015-07-06:
Hi jihoon.

I've just pushed an update to the repository that fixes several issues with the MATLAB example script. Most importantly, the example script was set up to connect to XPC on port 49005, when it should have been connecting on port 49009. Could you double check your connection settings and see if changing `openUDP` to connect on 49009 resolves the issue?

Thanks,
Jason

jihoon7942, 2015-07-08:
Hi Jason.

As you fixed and updated, MATLAB example script had some minor errors aboutUDP port. But it seems that's not the problem I have. I've been using your mfile functions without any problem after some minor modification in UDP port just like your new example.

But I'm using Simulink platform to communicate with X-Plane 10 via UDP, and I cannot find any better way than using the "MATLAB Interpreted function" block which significantly slow down the speed of the whole simulation and even repeatedly cause errors during the simulation. Could you possibly offer an example Simulink code that describe how to use your mfile functions within Simulink? 

Thanks,
Jihoon

jason-watkins, 2015-07-08:
Hi Jihoon.

I just wanted to let you know that I will be looking into the Simulink issues when I get to work tomorrow.

Unfortunately, I don't have any immediate answers for you. Neither I nor the other developer for X-Plane Connect have much experience with Simulink. I will ask around tomorrow and do my best to get someone with more Simulink knowledge to take a look at the code.

Would you be able to post your code on GitHub or email it to me? A reduced example that just demonstrates basic operation of the plugin would be fine. I haven't had a chance to try to get the code working in Simulink at all, and it would be quicker if I could use your code to demonstrate the Simulink issues rather than spending a couple hours writing up a demo myself.

-Jason

jihoon7942, 2015-07-08:
Hi Jason,

I'm going to send you a reduced example that demonstrates basic operation of the plugin in Simulink by email then.

Thanks,
Jihoon

jason-watkins, 2015-07-08:
Thanks Jihoon. My email is on my profile, but just in case it's jason.w.watkins@nasa.gov

-Jason

jason-watkins, 2015-07-27:
Based on offline discussions, I believe that we have identified that the core issue is not related to SimuLink in particular, but rather that the plugin was being overwhelmed with requests and was not able to keep up.

closed 2015-07-27 by jason-watkins