nasa/XPlaneConnect #79: Merge / cooperate with ExtPlane

vranki, 2015-08-28:
Hi,

I'm the author of ExtPlane plugin I just found XPlaneConnect. Looks like the feature set and target of the project is almost identical to ExtPlane's. XPlaneConnect seems to use a bit lower level UDP approach compared to ExtPlane's TCP socket, but still they do the same thing.

Would you be interested in merging the two projects into a single X-Plane comms plugin? I think it's a waste of effort to implement practically the same thing twice.

jason-watkins, 2015-09-06:
Hi Ville,

Unfortunately, I was the only person working on X-Plane Connect full time this summer, and I'm back at school now, so we may not have enough time to really address this in the near future.

That said, I agree that there's quite a bit of duplicated work, so I'll try to take a look at ExtPlane when I have time and see how we can integrate the two plugins. Do you have a suggested plan for integrating them?

Two issues I see right away:

 - The use of UDP to maximize throughput is one of the key requirements we had for the internal clients we originally wrote X-Plane Connect for. It should be relatively easy for the project to support both TCP and UDP on separate ports though.

- The GPL license may not play nice with NASA's internal requirements. @teubert will have to check on that, but I don't think we can release our code under GPL, which if my understanding of GPL is correct means we can't merge any of your code into our repo under the current license.

teubert, 2015-09-08:
Hello Ville,

Thank you for your message.

I also agree that there is some duplicated work. Like @jason-watkins<https://github.com/jason-watkins> mentioned the GPL license may not fit with the NASA open source software requirements. I'll look into this and get back to you.

I would definitely like to pursue this further though. This could be a good opportunity to improve our tools.

Thank you again,
Chris

----
Christopher Teubert
SGT, Inc.
NASA Ames Research Center

closed 2016-06-29 by jason-watkins