nasa/XPlaneConnect #80: Unititialized variable causes error in xpcExample

TurboTiger, 2015-09-08:
In *main.cpp* from xpcExample (found in: XPlaneConnect-master\C\xpcExample\xpcExample\src), the variable **int tSize** is uninitialized.  This causes the following error to be returned in the console:

[image]

The following error is also returned by visual studio:

[image]

This bug can be fixed by initializing the variable as follows:

    int tSize=1;

Cheers!

jason-watkins, 2015-09-08:
Hi TurboTiger.

`tSize` is uninitialized on purpose because it is only an allocation. The address of `tSize` is passed into `getDREF`, which will set its value. It seems like there is something else going on on your machine.

This doesn't cause any errors on either of my Windows machines (or on Mac as far as I know, but I don't test the Mac build myself). Could you share some additional information about the machine you're running on? In particular, I would be interested in your operating system, Visual Studio version, and basic hardware specs.

If you could also let me know exactly what line the access violation occurs on, that would greatly help in narrowing down your issue.

Thanks,
Jason

jason-watkins, 2015-09-08:
Hi TurboTiger.

I'm not sure why, but this bug didn't cause any problems for me on my machine. After tracing through the code though, I agree with your proposed solution. I've made the relevant change and committed it.

Thanks,
Jason

closed 2015-09-08 by jason-watkins