nasa/XPlaneConnect #82: Matlab readData Bug

oberion, 2016-01-26:
Hi it seems that in Matlab the readDATA function is not working properly. I always get the multiple clients issue although the global clients variable has only one entry. This can be fixed by changing:
assert(istrue(... -> assert(isequal(...
Further Matlab wants to call socket.readDATA() although the function should be readData(). 
But still with this fixes, the code does not run and there seems to be a problem for me with the buffer when reading out the data. This seems to be within the java code.

jason-watkins, 2016-01-26:
Hi oberion,

Thanks for the report. If you would like to submit a pull request with the fixes you've outlined above, we will be happy to review them and accept the pull request.

For the remaining Java issues, I will take a look at the Java code when I have a chance and see if I can identify the error. XPlaneConnect is no longer my full time job, so it will be this evening (Pacific Time) at the earliest before I can take a look. (@teubert might be able to get too it earlier)

In the mean time, would you be able to post a snippet demonstrating how you are using the `readDATA` function? That would help us make sure that the function works well for your specific use case.

Best,
Jason Watkins

oberion, 2016-01-27:
Hi,
Sorry, since im in europe i have a different time than you, so i didn't see your reply until now. Thanks for taking a look at it. 
So the first error i get is simply that MATLAB does not know the istrue() function (MATLAB 2015a) so i had to replace it with the isequal function. When i have done this i get the following error:
```
No appropriate method, property, or field 'readDATA' for class 'gov.nasa.xpc.XPlaneConnect'.

Error in XPlaneConnect.readDATA (line 49)
result.raw = socket.readDATA();

Error in XPCTest2 (line 34)
result = readDATA()
```
 Which i can fix with the second change mentioned above (see the pullrequest if done). But then is still get this error:
```
java.lang.IndexOutOfBoundsException

	at java.nio.Buffer.checkIndex(Unknown Source)

	at java.nio.HeapByteBuffer.get(Unknown Source)

	at gov.nasa.xpc.XPlaneConnect.readData(XPlaneConnect.java:678)

Error in XPCTest2 (line 34)
result = readDATA()
```
About my matlab code, i only did a few test until now to check the API so the code is fairly simple till now.
```
clear variables
%clear global
close all
fclose all;
clc

%% Import the functions
addpath('../XPlaneConnect/MATLAB');
import XPlaneConnect.*

%% Input parameters
hostAddr = '192.168.0.4';

%% Close old open clients
global clients
for i = 1:length(clients)
    closeUDP(clients(i))
end
   
%% Open the Connection
socket = openUDP(hostAddr);

%% select the Datastreams
selectDATA([1 3 4 16:21]);

%% read the Data
result = readDATA();

%% Close the Connection
closeUDP(socket);
```
I also have seen that i can get to the data vie the getDREFs command as an alternative.

jason-watkins, 2016-01-27:
Hey oberion,

Thanks for the update and code samples. I would highly recommend using `getDREFs` in place of `readDATA` if you can. `getDREFs` is much better tested and getting/setting X-Plane data via DREF is much more robust than using the DATA interface in our experience.

I'll go ahead and leave this issue open since there is a bug to be fixed, but I'm going to assume that you're able to get what you're after via `getDREFs`. If it turns out you do need the `readDATA` function, please let me know and I'll make it a priority again to resolve this issue.

Best,
Jason

closed 2016-06-20 by jason-watkins