nasa/XPlaneConnect #85: New Feature: Faster getting and setting of DREFs

jason-watkins, 2016-03-02:
Related to the discussion on #51, I think it might be useful to facilitate more efficient getting and setting of data refs. I'm envisioning the following new commands:

**Get DREF Pointer**: Takes a dataref string, and returns the internal pointer used by the X-Plane API. This would allow users to cache pointers to frequently used datarefs and only send those pointers rather than sending the full dref string each time they need to get or set a value. The plugin already caches these pointers internally, but having clients send pointers over the network could significantly cut down on the amount of data that the plugin needs to process in order to retrieve the right pointer.

**Get DREF(s) (pointer)**: Identical to the existing GetDREF(s) command, but pass pointers instead of the dref names.

**Set DREF (pointer)**: Identical to the existing SetDREF command, but pass a pointer instead of the dref name.

TurboTiger, 2016-03-03:
Sounds good!  Might it also be possible to implement a feature to continuously send a selected set of datarefs on each flight loop without having to request them each time?

jason-watkins, 2016-03-04:
The current architecture of XPC doesn't include a good mechanism for the server to "push" updates to a client, but I'll keep the idea in mind and see what I can come up with.