nasa/XPlaneConnect #87: Bugfix for sendDREFs example code

TurboTiger, 2016-03-04:
There are a number of bugs in the example code for the sendDREFs() function:

-  `char* drefs[]` should be `const char* drefs[]`

- semicolon missing after `malloc(4)`

- `sendDREFs()` requires a fifth input parameter: `int count`, which represents the number of datarefs being sent

- The second instance of `values[0][0]` should be `values[1][0]` to refer to the value for the second dataref.

I have incorporated the above bugfixes into a new, expanded example, included below.  Feel free to pare it down if desired and use it on the XPC website.  Note that the datarefs being set have been changed to the taxi and beacon lights.

```
const char* XPlane_IP = "127.0.0.1"; //IP address of PC running X-Plane
XPCSocket sock = openUDP(XPlane_IP);

int err;	//holds return code
const char* drefs[] = { "sim/cockpit2/switches/landing_lights_on", "sim/cockpit2/switches/beacon_on" };
float* values[2]; //A multidimensional array of values representing the data to set.

for (int i = 0; i < 2; ++i)
{
	values[i] = (float*)malloc(4);
}
//The following may be used in place of the FOR loop above if desired
//values[0] = (float*)malloc(1 * sizeof(float));
//values[1] = (float*)malloc(1 * sizeof(float));

//Note: with XPC, everything is sent as floats, even if X-Plane wants an int.
values[0][0] = 0.0F; //turn off landing lights (1.0F = ON)
values[1][0] = 0.0F; //turn off beacon (1.0F = ON)
int sizes[] = { 1, 1 }; // The number of items in each row of "values"
int count = 2; //The number of datarefs being sent
	
err = sendDREFs(sock, drefs, values, sizes, count);
	
switch (err)  //interpret return value (error code)
{
case -1:
	std::cout << "\nERROR: Length of dref is greater than 255 characters." << "\n";
	break;
case -2:
	std::cout << "\nERROR: There are more than 255 items in \"value\"." << "\n";
	break;
case -3:
	std::cout << "\nERROR: XPC client unable to send command." << "\n";
	break;
case 0:
	std::cout << "\nSUCCESS: XPC sent the specified datarefs." << "\n\n";
}
```

jason-watkins, 2016-03-04:
Hey Brent,

Thanks for catching those errors. I've fixed up the errors you pointed out and added your code to the `sendDREFs` wiki page as well.

Best,
Jason

closed 2016-03-04 by jason-watkins

cwnsoftware, 2017-06-22:
Hey Brent,

In X-Plane11, I can't control the beacon lights of the JarDesign A320 aircraft with the above code. What's the problem?

Best,
Ken

jason-watkins, 2017-06-22:
Hi Kent,

Can you confirm that you are able to control the beacon lights of aircraft that come with X-Plane? Assuming that's the case I suspect the issue is with the JarDesign aircraft. I have never added a custom aircraft to X-Plane, so the creators of the aircraft would likely have a better idea of what's going on that I do.

If you are unable to change the beacon lights for any aircraft, please open a new ticket instead of replying to this one.

Thanks,
Jason