nasa/XPlaneConnect #89: Matlab UDP connection problem

Endf22, 2016-03-27:
Hello everyone,
I m trying to connect with xplane from matlab environment, but i have some problems about connecting xplane via UDP.
I want to pause xplane from matlab but nothing changes.
My code;
`Socket = openUDP('127.0.0.1',49003,49001);`

Xplane Connection Settings
[image]

What's wrong with this connection?  Can you help me with this?

Any help appreciated, thank you;)

teubert, 2016-03-27:
Hello!

Thank you for contacting us with your problems. Let’s see what I can do to clear this up.

The screen you’re looking at is for the x-plane UDP interface. The XPlaneConnect tool bypasses this interface using our own plugin. To use X-PlaneConnect you should be sending your messages to the xpcPlugin (at 49009).

So here’s what I suggest you try:
1. If you haven’t already, copy the X-Plane Plugin folder (xpcPlugin/XPlaneConnect) to the plugin directory ([X-Plane Directory]/Resources/plugins/)
2. Replace your openUDP command with the following line (this will use the default settings):
Socket = openUDP();
3. Restart X-Plane and try it out

Let me know if this works for you. If not I would be happy to help you debug this further.

Thank you,
Chris

——
Christopher Teubert
Research Engineer, SGT Inc
NASA Ames Research Center
Christopher.A.Teubert@nasa.gov<mailto:Christopher.A.Teubert@nasa.gov>

[udp]<https://cloud.githubusercontent.com/assets/18101862/14066227/9e6978b6-f44c-11e5-992f-95f69513699a.JPG>

Endf22, 2016-03-28:
Thank you for the answer,
 I just thought that those UDP ports which are shown in Xplane connection settings are to be used. However, xpc plugin port (49000) should be used.
There is no problem now, I fixed it with `Socket = openUDP();`, no need to put there any ports

closed 2016-03-28 by Endf22

teubert, 2016-03-28:
Great, I'm glad it worked!