nasa/XPlaneConnect #90: send "Char" command to Xplane

Endf22, 2016-03-28:
Is it possible to send char commands to Xplane by using this plugin? 
For example, the simulation can be paused by using keyboard command "p" in Xplane.  
Is there any plugin class to send string to this plugin in default?   I havent found the related structure in this plugin project or am i missing something?

jason-watkins, 2016-03-28:
Hi Endf,

X-Plane Connect does not currently have a command for sending raw key data to X-Plane. The underlying X-Plane Plugin SDK does have a function to allow this, but it is marked as deprecated. Most functionality in X-Plane can be controlled by setting Datarefs instead. For other functionality, we prefer to expose that functionality as a separate function.

For the example you mentioned, we have the [pauseSim](https://github.com/nasa/XPlaneConnect/wiki/pauseSim) command, which pauses the flight simulation (although it doesn't actually pause the application in the same way that pressing <kbd>p</kbd> does).

Endf22, 2016-03-28:
Hi Mr. Watkins,

I am currently using that command "pauseSim" in order to stop the Xplane physics engine. I just wanted to make sure that whether X-Plane Connect plugin sends raw key data to Xplane or not. 

Thank you for the quick reply,

closed 2016-03-28 by Endf22