nasa/XPlaneConnect #91: Sudden aircraft crash after "pauseSim" command

Endf22, 2016-03-29:
Hi folks, 

There is a sudden aircraft crash after using pauseSim(0,Socket). After that, aircraft has anormal attitude For example, pitch deg goes to 80 deg suddenly and then other attitudes become useless. 
Is there anyone who has any idea?

My code is simple as in matlab example file in Xplane Connect 
[Xpconnect.txt](https://github.com/nasa/XPlaneConnect/files/193448/Xpconnect.txt)

Thank you,

teubert, 2016-03-29:
Hello Endf22,

This is interesting. The cause is not immediately obvious to me. I'll try your script and get back to you later today.

In the meantime could you attach the XPCLog file created by the plugin. It should be in the same directory as the Plugin.

Thanks,
Chris

Endf22, 2016-03-29:
Hello,
I am trying to understand the reason why aircraft behaves like that. 
I sent the XPCLog file
[XPCLog.txt](https://github.com/nasa/XPlaneConnect/files/194083/XPCLog.txt)

teubert, 2016-03-30:
Hello again Endf22,

I ran your scripts and I'm getting exactly the same behavior as you described. I reran the Matlab Example script  and I get this behavior as well. I now remember this being an unanswered issue where the transition between paused and unpaused can be violent. I'm going to leave this open and continue to investigate.

A few possible causes come to mind:
1. There is another control/state variable that is not set that is acting on the plane
2. After resuming it takes a timestep for the changes to take affect.

@jason-watkins Do you have anything else to add here?

I want to get to the bottom of this though. Let me know if you discover anything in your investigations.

Sorry I can't be more help at this point.

Thanks,
Chris

Endf22, 2016-03-30:
Hi again,
I have found this info about the transitioning from disabled to enabled flight model.
http://www.xsquawkbox.net/xpsdk/mediawiki/MovingThePlane#Transitioning_From_Disabled_to_Enabled_Flight_Model 
In that link it says velocity vector and the quaternion q should be calculated in order to match with other flight control parameters such as angular rotation variables (p,q,r).

The other problem I have found so far in the plugin is that sendPOSI command is not working as expected. The altitude and the position inputs dont match with the outputs. For example altitude input is 1500 ft. (alt_msl) but it moves aircraft to somewhere else and then A/C crashes.

Thank you,

jason-watkins, 2016-03-30:
Thanks for the research Enf!

I don't have my PC with me today, but I'll take a look and make sure our code conforms to the information provided in your link when I get a chance.

The sendPOSI command should work, but there are a lot of edge cases. Could you post the approximate location of the airplane before sending the command along withe the coordinates you are moving it to?

Chris, I remember this issue as well. In fact, I thought I had fixed it, but I can't find a matching issue so maybe not. It looks like the link above addresses the issue directly, so it should be a straightforward fix.

Endf22, 2016-03-30:
Hi,
The approximate location that
Airport: KSAN (San Diego Intl)
Sİtuation: Approach 10nm
lat: 37.78
long:-117.4
alt:1500 ft. (alt_msl)

jason-watkins, 2016-06-20:
Finally found time to review this. We were not resetting the speed and orientation. I've added code to update the orientation. Updating velocity is going to take significantly more work, and I'm not entirely sure that it's actually necessary for normal use.

commit 35285cac (2016-06-20): Updated SetPosition

 - SetPosition now updates the aircraft's orientation after moving the aircraft
 - This should improve the handling of the aircraft when setting the position while paused and then unpausing
 - Fixes #91
closed 2017-05-03 by teubert