nasa/XPlaneConnect #92: openUDP error when running example.m

mattpilm, 2016-04-03:
Hi,

I've downloaded XPlaneConnect, and copied the appropriate folder into the plugins folder of xplane (10.45 64bit). I can see the plugin in xplane and have ensured that it is enabled. When I try to run the example code in matlab (2012b) I get the following error.

[image]

Am I missing some network configuration steps in matlab or xplane? Apologies in advance if I'm doing something fairly obvious wrong. This is not my area of expertise. Any assistance would be very much appreciated.

With very kind regards

Matt

Endf22, 2016-04-08:
Hello,

Try this `Socket = openUDP();`

Kind Regards,

mattpilm, 2016-04-08:
Many thanks for the suggestion,

I tried that both in example.m and at the command line and get the same errors unfortunately. I've reinstalled Java and still no luck...

[image]

Best

Matt

teubert, 2016-04-08:
Hello mattpilm,

Thank you for your patience so far. I’m not familiar with this error, and I’m not sure why it would be happening. I so far haven’t been able to recreate it.

Do you have the file XPlaneConnect.jar in your +XPlaneConnect folder? My intuition first tells me that I somehow the matlab function cannot access the XPlaneConnect class in the XPlaneConnect.jar file. That said there is a check on line 33 that should check for that. 

If the XPlaneConnect.jar file is present could you try redownloading it from github. It’s possible that the jar file is corrupted somehow.

I’ll keep thinking about this. Unfortunately my experience with matlab is limited, but I will try to pull in some of my colleagues that are more familiar with this.

Good luck!
——
Christopher Teubert
Research Engineer, SGT Inc
NASA Ames Research Center
Christopher.A.Teubert@nasa.gov

mattpilm, 2016-04-08:
Hi Christopher,

Thanks for your response. My intuition pointed me to the .jar file also hence the java re-install. The .jar file is present in the folder so I'll try downloading it again and see if that helps. I'll report back as soon as I can.

With very kind regards, and thanks again,

Matt

mattpilm, 2016-04-08:
Hi Christopher,

I identified a path problem. Matlab didn't like the filename +XPlaneConnect when adding to the path so I deleted the + and Matlab now acknowledges that the functions exist.

I also tried to bring the java class into the Matlab workspace using the instructions here:
http://uk.mathworks.com/help/matlab/matlab_external/bringing-java-classes-and-methods-into-matlab-workspace.html

I now get the following errors:

Warning: A Java exception occurred trying to load the gov/nasa/xpc/XPlaneConnect
class:
Java exception occurred:
java.lang.UnsupportedClassVersionError: gov/nasa/xpc/XPlaneConnect : Unsupported
major.minor version 51.0

	at java.lang.ClassLoader.defineClass1(Native Method)

	at java.lang.ClassLoader.defineClass(Unknown Source)

	at java.security.SecureClassLoader.defineClass(Unknown Source)

	at java.net.URLClassLoader.defineClass(Unknown Source)

	at java.net.URLClassLoader.access$000(Unknown Source)

	at java.net.URLClassLoader$1.run(Unknown Source)

	at java.security.AccessController.doPrivileged(Native Method)

	at java.net.URLClassLoader.findClass(Unknown Source)

	at java.lang.ClassLoader.loadClass(Unknown Source)

	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)

	at java.lang.ClassLoader.loadClass(Unknown Source)

	at java.lang.ClassLoader.loadClassInternal(Unknown Source)

	at java.lang.Class.forName0(Native Method)

	at java.lang.Class.forName(Unknown Source)

	at com.mathworks.jmi.OpaqueJavaInterface.findClass(OpaqueJavaInterface.java:470)
 
> In Playback at 15 
Warning: Invalid file or directory
'C:\Users\Matt\Documents\MATLAB\XPlaneConnect.jar'. 
> In javaclasspath>local_validate_dynamic_path at 274
  In javaclasspath>local_javapath at 182
  In javaclasspath at 119
  In javaaddpath at 71
  In openUDP at 35
  In Playback at 15 
Warning: A Java exception occurred trying to load the gov/nasa/xpc/XPlaneConnect
class:
Java exception occurred:
java.lang.UnsupportedClassVersionError: gov/nasa/xpc/XPlaneConnect : Unsupported
major.minor version 51.0

	at java.lang.ClassLoader.defineClass1(Native Method)

	at java.lang.ClassLoader.defineClass(Unknown Source)

	at java.security.SecureClassLoader.defineClass(Unknown Source)

	at java.net.URLClassLoader.defineClass(Unknown Source)

	at java.net.URLClassLoader.access$000(Unknown Source)

	at java.net.URLClassLoader$1.run(Unknown Source)

	at java.security.AccessController.doPrivileged(Native Method)

	at java.net.URLClassLoader.findClass(Unknown Source)

	at java.lang.ClassLoader.loadClass(Unknown Source)

	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)

	at java.lang.ClassLoader.loadClass(Unknown Source)

	at java.lang.ClassLoader.loadClassInternal(Unknown Source)

	at java.lang.Class.forName0(Native Method)

	at java.lang.Class.forName(Unknown Source)

	at
    com.mathworks.jmi.OpaqueJavaInterface.findClass(OpaqueJavaInterface.java:470)
 
> In openUDP at 37
  In Playback at 15 
Undefined variable "gov" or class "gov.nasa.xpc.XPlaneConnect".

Error in openUDP (line 37)
socket = gov.nasa.xpc.XPlaneConnect(p.Results.xpHost, p.Results.xpPort,
p.Results.port, p.Results.timeout);

Error in Playback (line 15)
Socket = openUDP(); % Open connection to X-Plane

duguangxun, 2016-05-05:
It is because of the version of java being used in your Matlab and in your java code. Make sure they are using the same version. For example I was compiling my code using jdk 1.6 while my Matlab was using jdk 1.5. I changed my main java to 1.5 and then Matlab was able to load the jar file. I am using MATLAB 2011b which uses Java 1.6.0_17-b04. And my MATLAB failed to import the file XPlaneConnect.jar. So the problem is that what is the exact java release number when you compile the java files in this project?

jason-watkins, 2016-06-20:
Sorry for not responding to this earlier. This issue slipped by me completely when it was opened.

The current jar file distributed with the MATLAB client is compiled targeting Java 7. I'm not sure if we even have access to MATLAB 2011 anymore, so it's probably not practical for us to support XPC on MATLAB 2011. The .jar file is just the compiled Java client though, so as a workaround you should be able to recompile the XPC Java client on your system and use that to replace the `XPlaneConnect.jar` that we include in the repo.

closed 2016-06-20 by jason-watkins

mattpilm, 2016-06-21:
Thanks for all your help...

Best