nasa/XPlaneConnect #94: readDATA() always fails to read from socket

TurboTiger, 2016-06-10:
I am trying to use the readDATA() function to receive two UDP datasets selected for "Internet" output in X-Plane, however, the function always returns "ERROR: Failed to read from socket."  My code is included below:

```
#include <stdio.h>
#include "xplaneConnect.h"

int main()
{
	printf("XPlaneConnect Example: readDATA()\n\n");

	// Open Socket
	const char* IP = "127.0.0.1";      //IP Address of computer running X-Plane
	XPCSocket sock = openUDP(IP);

	// Read 2 rows of data
	const int rows = 2;
	float data[rows][9];
	readDATA(sock, data, rows);

	system("PAUSE");

	closeUDP(sock);

	return 0;
}
```

Thanks!!

jason-watkins, 2016-06-10:
Hey Turbo,

We don't use the sendDATA/readDATA functions much internally, so they aren't very well tested. I'll take a look and see if I can reproduce the error. If you don't here from me by Monday morning (Pacific time) please drop me a note here to remind me.

Thanks,
Jason

TurboTiger, 2016-06-10:
I think I just solved my own problem.  Details to follow.

TurboTiger, 2016-06-10:
Thanks for the super-fast reply @jason-watkins!  The problem was due to incorrect IP address settings.  It seems that in order to receive X-Plane UDP data, the port number to which X-Plane is sending datagrams must be _explicitly specified_.

Instead of using:

```C
const char* IP = "127.0.0.1";      //IP Address of computer running X-Plane
XPCSocket sock = openUDP(IP);
```

it was necessary to use this:

```C
const char* xpIP = "127.0.0.1";            //IP Address of computer running X-Plane
	unsigned short xpPort = 49007;   //default port number XPC listens on
	unsigned short port = 49003;	     //port number to which X-Plane is set to send UDP packets
	XPCSocket sock = aopenUDP(xpIP, xpPort, port);
```

**Note that the [documentation](https://github.com/nasa/XPlaneConnect/wiki/openUDP) regarding the number of arguments required by `openUDP()` and `aopenUDP()` appears to be outdated.**

My complete working example code appears below.  Feel free to use it on the wiki.

```C++
#include <stdio.h>
#include "xplaneConnect.h"

int main()
{
	printf("XPlaneConnect Example: readDATA()\n\n");

	// Open Socket
	const char* xpIP = "127.0.0.1"; //IP Address of computer running X-Plane
	unsigned short xpPort = 49007;	//default port number XPC listens on
	unsigned short port = 49003;	//port number to which X-Plane is set to send UDP packets
	XPCSocket sock = aopenUDP(xpIP, xpPort, port); //docs seem to be outdated on these...

	// Read 2 rows of data
	const int rows = 2;
	float data[rows][9]; //data[0] is dataset index number, data[1] to data[9] are the contents of the dataset
	readDATA(sock, data, rows);

	printf("Received UDP datasets %.0f and %.0f, consisting of: \n", data[0][0], data[1][0]);

	//print contents of first dataset
	for (int i = 1; i <= 8; i++)
	{
		printf("%f,", data[0][i]);
	}
	printf("\n");
	
	//print contents of second dataset
	for (int i = 1; i <= 8; i++)
	{
		printf("%f,", data[1][i]);
	}
	printf("\n");

	//keep console open on Windows
	system("PAUSE");
	
	closeUDP(sock);
	
	return 0;
}

```

jason-watkins, 2016-06-10:
Hey Turbo,

Thanks for the update. I'm glad you got things working, but I think there might still be a bug in the `readDATA` function. You shouldn't need do set the ports explicitly. I'll take a look on Monday anyway.

Also thanks for catching the Wiki error, I've corrected that as well.

vincedjango, 2018-05-26:
Hello,
Just to let you know that I tried  TurboTiger solution and it didnt work, I still get the same error.

Seromonto, 2019-02-13:
What port i need set in DataRefs in Xplane 11? The same as IP in General output?

And what ports i need set in NETWORK tab in xplane 11?

jason-watkins, 2019-02-13:
Hi Seromonto, if you have a question, please create a new issue. X-Plane Connect only uses one port, so if readData is working for you I don't think it's a port issue.

vtoltrianee, 2020-03-20:
hi ,
i'm new to x plane connect( Xpc ) tool box , i want to communicate X plane 11 with C-code via UDP ,if you have any basic example in c -code  mention here and also if you know how to do that please mention the step by step procedure 
any answers are most welcome
Thanks.

RakeshVangaveti, 2020-04-30:
> Thanks for the super-fast reply @jason-watkins! The problem was due to incorrect IP address settings. It seems that in order to receive X-Plane UDP data, the port number to which X-Plane is sending [...]

hello @TurboTiger,as mentions  xplane-udp recieve program how did you tested that? readDATA() , i'm also looking for that , to test read and send data to xplane i've gone through nasa/xplaneconnect/wiki documentation got some confusion ,help me the communication between xplane and send/recieve programs using sendDATA()/readDATA()
thanks in advance ...
hope you got it.

praveenethiraj, 2021-01-19:
@TurboTiger Hello! Thank you for this piece of code and for the explanation. I know I am asking a question on this topic after so long but I was recently trying to run it and Visual Studio threw me a "Constant expression expected" at the line
float data[rows][9]

Google tells me that there might be a problem with the memory allocation and that a pointer might help me but I keep hitting dead ends. 

Could you let me know if you had any minor corrections to make in the final version please?

rohanblueboybaijal, 2021-06-06:
Hi
@TurboTiger @jason-watkins 
I've been having similar issues with the Python code when I use `readDATA()`.
I tried setting the port to **49003** and the `readDATA()` function worked pretty smoothly for me.

I then tried running the other sample codes given (monitorExample.py) and noticed they started throwing errors (mainly "unexpected response length"). Then I put the `readDATA()` function there (even if I don't need it) and it started working fine again.

Perhaps this is also somewhat related to the datarefs issue raised by @Seromonto.  So, if we're playing around with the settings to get the data, I think we'll have to be careful and ensure there's one `readDATA()` call in the loop too.

ericjhkim, 2021-08-12:
I'm having a similar issue where readData() does not work. The readUDP() function at 16384 always returns a time out error. I've tried changing the port number in X-Plane and changing 16384 to match, but I'm continuing to get the same error.
Perhaps others have encountered something similar?