nasa/XPlaneConnect #95: "java.io.IOException :No response received"

HunterOmega, 2016-06-20:
I am using XPlaneConnect in Matlab, and after a random time (ranging from 3 to 120 seconds so far) the monitor example (unmodified) provided gives the error:
[image]

I'm using Matlab R2015a and X-Plane 10.45

Thanks.

jason-watkins, 2016-06-20:
Hi Hunter,

Are you running the Matalb client and X-Plane on the same computer? Can you also upload your XPC log (`XPCLog.txt` in the X-Plane install directory)?

HunterOmega, 2016-06-21:
Both Matlab and X-Plane are running on the same computer. XPC log attached.

[XPCLog.txt](https://github.com/nasa/XPlaneConnect/files/325110/XPCLog.txt)

HunterOmega, 2016-06-22:
It would appear that a program called "Target GUI" which is used to change the inputs generated by Thrustmaster products (I'm using the Warthog HOTAS) caused the above error to occur.

It was configured to run in the background, which is why I didn't disable it before.

The software is freely available at [http://ts.thrustmaster.com/eng/index.php?pg=view_files&gid=1&fid=3&pid=352&cid=5](url) should you wish to accommodate for it.

Thanks,
Hunter

HunterOmega, 2016-07-06:
Also tried running both X-Plane and Matlab in and out of administrator mode, but the problem still occurred.

jason-watkins, 2016-07-06:
Hi Hunter,

Unfortunately since I can't reproduce the problem, it's difficult for me to offer specific advice. One thing you might try is increasing the timeout (by calling `openUDP('127.0.0.1', 49009, 0, 1000)`, where "1000" is the timeout in milliseconds. It defaults to 100ms, which should be more than enough for localhost communications, but you never know what might be delaying things inside the kernel.

I did notice one odd thing about the log file you posted. It ends with the socket logging a send, but there should be some additional lines describing the shutdown process when X-Plane closes. Is X-Plane crashing on you at all?

HunterOmega, 2016-07-06:
It rarely crashes, but that was a normal shutdown. Will try that UDP setting.

HunterOmega, 2016-07-14:
Had to increase the timeout to 10 secs, but it's working now. Seems that the problem was cause by X-Plane not wanting to run smoothly, and when the frame rate dropped temporarily, it dropped the connection.

MichaelRThompson, 2017-06-20:
I am having the same issue running XPlaneConnect in MATLAB.  I can get the monitor example script to run by increasing the timeout to 10sec like HunterOmega suggested.

However, in creating my own scripts to use X-Plane to visualize an external control system, I constantly come across this error if I attempt to update the position of an aircraft at a step time of less than 0.01 seconds.  Eventually the script breaks, and I cannot connect to X-Plane at all with MATLAB until I close and restart the program.

Looking at XPCLog.txt, it always ends with ERROR: Failed to bind socket. (Error code 10048)

jason-watkins, 2017-06-22:
Hi Michael,

100Hz is a very high update rate. I am not surprised that it doesn't work reliably. Unless X-Plane is running at greater than 100 FPS (Which would be quite a feat) you're forcing X-Plane to update the plane's position multiple times per frame. Since physics calculations also occur once per frame, I can't imagine what purpose that would serve. 

Also keep in mind that for simplicity's sake, the XPC plugin runs synchronously inside the main X-Plane UI thread. All of that extra message processing in XPC could have a detrimental effect on X-Plane's frame rate.

If you really need to update that fast, feel free to open a new issue detailing the use case and I'll see if there's anything I can do to support it. I'm going to close this issue though since I don't see anything that needs work here.

closed 2017-06-22 by jason-watkins

arcibiyo, 2020-10-05:
I have a same error when ı send data and get data together. However, ı can just send data to xplane or get data from xplane. How can ı solve this problem ? Probably, ı need to use different udp port for this situation but ı can not find solution. I am waiting answer from you. Thanks..

 ``>> GRD
Loc: (40.996220, 28.806032, 0.000000)
Error using XPlaneConnect.getPOSI (line 27)
Java exception occurred:
**java.io.IOException: No response received.**

	at gov.nasa.xpc.XPlaneConnect.**getPOSI**(XPlaneConnect.java:565)

Error in GRD (line 11)
        posi = getPOSI(0, Socketx);``