nasa/XPlaneConnect #97: readDATA needs a better return value

Bob-King, 2016-07-10:
Hi, I'm using readDATA to get data, but find its return value doesn't meet my expectation.

So far, the function readDATA returns the following values

1. -1 on failure

2. the rows read if the rows read exceeds the rows specified

3. the rows specified if the rows read doesn't exceed the rows specified

The caller can't differentiate from 2 and 3, and can't know the rows read if there is less rows available. I think this function had better always return the rows read from xplane while not changing current copy logic. Then the caller can makes their own decision based on the return value.

Bob-King, 2016-07-10:
I just notice that buffer overflow will occur on copying if the rows read exceeds the rows specified.

jason-watkins, 2016-07-10:
Hi Bob-King,

I agree that my error checking in `readDATA` seems to be a bit odd. I've pushed a patch so that the function should now always return either -1 or the number of rows written to `data`, which was the original intent. This incidentally fixes the buffer overflow issue as well.

Bob-King, 2016-07-10:
Hi Jason:

I see your change. It's fine although it is not my preference:)

Thanks

commit 5d23d482 (2016-07-10): Fixed incorrect error checking in readDATA

 - Resolves #97
closed 2016-07-10 by Bob-King