nasa/XPlaneConnect #101: Timeout when using readDATA

chrismas11, 2017-03-03:
I'm trying to use the readDATA function in python, however the program always returns timeout - I've tried timeout values up to 100 seconds. The python programs work fine when using getPOSI or getCTRL (as per the monitorExample.py function), however when I replace getPOSI and getCTRL with readDATA it returns timeout. Is this just due to a lack of computing power? I have reduced the graphics on X-Plane to minimum, and have limited the data being sent over udp by X-Plane to pitch/roll/headings and angular velocities.

chrismas11, 2017-03-03:
Fixed the issue - the xpc python code as it comes in the download doesn't specify a value for port. Upon setting this to the port X-Plane is sending data to, readDATA works fine. Feeling noob :3

teubert, 2017-03-03:
I'm glad that you were able to fix your issue. Let me know if you have any other questions.

closed 2017-03-03 by teubert

chrismas11, 2017-03-03:
I am having one more issue, I'm not sure whether it's just due to a lack in my coding prowess...

The readDATA function is returning values, but the first element of each array is always 0, where it should be the reference to the type of data is being obtained. Have you come across this before?

rkalghatgi, 2017-05-09:
Hi chrismas11, would you be able to post where you set the port for readData? I am having the same issue. Thank you.