nasa/XPlaneConnect #102: Compile error on Linux

asweet42, 2017-04-24:
On compiling the C client for XPlaneConnect, on Linux, I got the following error:

[...]/xplaneConnect/C/src/xplaneConnect.c:125:17: error: storage size of ‘timeout’ isn’t known
  struct timeval timeout;

I resolved this by changing line 49 #include <time.h> to 

#ifdef _WIN32
  #include <time.h>
#else
  #include <sys/time.h>
#endif

That at least compiled on Linux and Windows. If there are configuration steps I didn't do, that would take care of this, please let me know. I didn't try running it.

jason-watkins, 2017-04-24:
Hi asweet,

Could you tell me what Linux distro and compiler you're using?

asweet42, 2017-04-24:
Hi Jason,

I'm Adam Sweet from Ames - I expected the note to go to Chris, I emailed
him about it.

but yes, I'm using Debian linux (in a VM actually) with gcc 4.9.2.

Adam

teubert, 2017-04-24:
Hi Adam, Jason,

A quick search reveals that timeval is defined in sys/time.h,  not time.h, which seems to be what is causing Adam's problem. I'm surprised that this hasn't been an issue earlier! I'll run some tests with the change.

Chris

jason-watkins, 2017-04-24:
@asweet42 I... should have put that together. In my defense I haven't had coffee yet.

@teubert Yeah, best guess it happened to work on the one setup I used to compile XPC, but not on all Linux setups. I have a feeling sys/time is "more correct" and will work on any *nix system.

jason-watkins, 2017-05-03:
Fixed in 1dfe7bf.

closed 2017-05-03 by jason-watkins