nasa/XPlaneConnect #109: Undefined Reference to openUDP

JacobKoetter, 2017-05-18:
Hello,

my plan is to set up a hardware-in-the-loop-simulation with using Xplane only for the visuals. The simulation is written in SIMULINK. So I thought that I just use the Matlab Function Block in SIMULINK and implement your provided MATLAB code. The only problem was that SIMULINK does not allow me to import java libraries.
So my next idea was that I write the code in C and then use an s-function for the implementation in SIMULINK. And here starts my problem:

So I just used the C Exmpale from the folder and put it into the s-function. Because of the knowledge that the matlab example script worked, I assumed the C scrip will also be working. And my main problem is now, that I get the following error:

Writing library for SP.mexw32
undefined reference to '_openUDP' 

+ all the other commands. So do you have any idea how to fix this? I think I'll have to combine the xplaneConnect.c and the xplaneConnect.h? But how?

I'll appreciate your answer.

Best regard
Jacob

JacobKoetter, 2017-05-19:
Hello,

a quick update. I forgot to compile xplaneconnect.c too. This solved the problem with the undefinded reference. The next problem is, that I get errors like:
xplaneConnect.c(103) : error C2275: 'WSADATA' : illegal use of
this type as an expression

But when I compile my script outside of MATLAB it's working. So is there maybe a problem with MATLAB?

jason-watkins, 2017-06-22:
Hi Jacob,

Sorry we weren't able to get to this sooner. If you're still having issues, could you post a minimal example that reproduces the problem?

Thanks,
Jason

jason-watkins, 2017-07-07:
Closed for lack of activity.

closed 2017-07-07 by jason-watkins