nasa/XPlaneConnect #110: Landing gear deploys automatically when stepping through points in MATLAB

MichaelRThompson, 2017-06-20:
I haven't been able to come up with an exact set of circumstances that this always happens, but it is a common issue that I run into.

I will be stepping an aircraft through a set of lat/lon/alt/orientation/gear values, and at every step, the aircraft seems to be attempting to deploy its landing gear.  Once the set of points is complete, the aircraft pauses in mid-air (as it is supposed to), but the landing gear deploys on its own.  This occurs with the physics sim turned on or off.

As an example:

`for n = 1:1000
    posi_1 = [19.6212513 + 0.001*n, -155.3792526, 10000, 0, 0, 0, 0];
    posi_2 = [19.6215513 + 0.001*n, -155.3792526, 10010, 0, 0, 0, 0];
    
    sendPOSI(posi_1, 0, Socket)
    sendPOSI(posi_2, 1, Socket)
    
    pause(0.01)
end`
This code should result in both aircraft completing the simulation with their gear stowed, but I run into the issue described above.

MichaelRThompson, 2017-06-21:
Looking in the log file, this is the line that is present when the gear value is set to 0.

09:33:00.169| INFO|DMAN|Setting gear (value:0.000000, immediate:1) for aircraft 0

jason-watkins, 2017-06-22:
Hi Michael,

The gear deploying as described does sound incorrect. I don't have time to test this today, but I will try to get to it tomorrow. In the mean time, you could try setting the gear value to either -1 or -9 instead of 0. This should prevent the plugin from touching the gear at all.

I believe the reason that `sendDREF({'sim/aircraft/parts/acf_gear_deploy'},0,Socket)` doesn't work is that the `sendPOSI` command is setting the gear handle via `sim/cockpit/switches/gear_handle_status`, whereas the `acf_gear_deploy` dref sets the actual position of the landing gear.

Best,
Jason

MichaelRThompson, 2017-06-23:
Thanks Jason, I can continue to look in the coming days as well.  I've tried the `gear_handle_status` dref, but don't see any change, even momentarily.  In testing, I think one way to usually ensure that the gear remains stowed is to take off within X-Plane before using the plugin.  I also know that you can't stow your gear when you are on the ground.  This makes me wonder if there is a flag within X-Plane where the plane still thinks there is a force being applied through the landing gear, and as a result, constantly tries to deploy it.

I imagine you could test this theory through a couple of the `sim/flightmodel/forces/` drefs.  I'll for sure get to that next week if you haven't by then.

jason-watkins, 2017-06-26:
Hi Michael,

Would it be possible for you to post your full code? (You can email it to me if you don't want share it publicly, email in my profile). I can't replicate your issue in my sim. No matter what I do, the gear follows `gear_handle_status` when physics are enabled, and does nothing unless I change `acf_gear_deploy` when physics are disabled.

-Jason

MichaelRThompson, 2017-06-29:
Here's some sample code I wrote up.  The route/orientation is nonsensical, but the issue remains.  Here's a quick summary of some of my findings.  I have three pauseSim commands in my code and have been switching up which ones are commented out.  And just for reference, I'm using the F22 that comes with X Plane 10 on Windows with MATLAB R2016b (although I can replicate it using multiple other a/c).

- Upon startup, if I run this code while the a/c is on the ground, and pause the sim before the for loop, the issue occurs exactly as I described it.

- If I run the code with "Pause command 1" commented out, the landing gear attempts to deploy for about half of the simulation, and then is stowed, and stays stowed.  Then, in further simulations, I am unable to replicate the issue.  

I've attached two plugin logs, one where the simulation is paused before the for loop, and one where it's not.

Simulation paused during for loop:
[XPCLog.txt](https://github.com/nasa/XPlaneConnect/files/1113106/XPCLog.txt)

Simulation unpaused during for loop:
[XPCLog.txt](https://github.com/nasa/XPlaneConnect/files/1113049/XPCLog.txt)

```
%% Troubleshooting landing gear issue

clear

addpath('../')
import XPlaneConnect.*

Socket = openUDP('127.0.0.1', 49009, 0, 10000);% Open connection to X-Plane

sendVIEW(Socket, gov.nasa.xpc.ViewType.Chase)

initial_posi = [36.8283486, -105.280669, 5000, 0, 0, 0, 0];
sendPOSI(initial_posi, 0, Socket)

% Unpause Command
pauseSim(0,Socket)

% Pause command 1
pauseSim(1, Socket)

for n = 1:1000
   posi = [36.8283486 + 0.01*sin(n*0.01), -105.280669 + 0.01*cos(n*0.01), 5000, 0, 0, 0, 0];
   
   sendPOSI(posi, 0, Socket);
   
   pause(0.01)
end

% Pause command 2
%pauseSim(1, Socket)
```

MichaelRThompson, 2017-07-12:
Note (mainly for myself later):  When I pause the sim on the runway and the move the a/c, the DataRef `sim/flightmodel/failures/onground_any` remains flagged.  It cannot be directly written, but could be the flag that is causing the a/c to refuse commands to stow the gear.

MichaelRThompson, 2017-07-15:
Continued investigation: Using the Data Ref Editor plugin, I can see that if I move the a/c directly from the ground while the sim is paused running the above code the DREF `sim/aircraft/parts/acf_gear_deploy` isn't actually a constant value once the code finishes.  It moves quickly back and forth between 1 and ~0.997.  This is despite the fact that `sim/cockpit/switches/gear_handle_status` stays constantly at 0.  The same behavior is seen with `sim/flightmodel2/gear/deploy_ratio`.  Always slightly decreasing from 1, but then being reset back to 1.

So obviously XPlane is reading the `gear_handle_status` and trying to set the appropriate DREFs to 0, but is conflicting with something else and immediately resetting the DREFs that actually display the deployed/stowed gear to 1.

Summary for me later:
`sim/cockpit2/controls/gear_handle_down` : Stays set at 0
`sim/cockpit/switches/gear_handle_status` : Stays set at 0
`sim/aircraft/parts/acf_gear_deploy` : Cycles between 1 and ~0.997
`sim/flightmodel2/gear/deploy_ratio` : Cycles between 1 and ~0.997
`sim/multiplayer/controls/gear_request` : Stays set at 0 for all active a/c
`sim/flightmodel/movingparts/[DREF for each gear]` : Cycles between 1 and ~0.997
`sim/aircraft/gear/acf_gear_retract` : Stays set at 1, meaning the gear is allowed to be retracted

`override_gearbrake` : No change with activation
`override_groundplane` : No change with activation

And I just went through every DREF with "gear" in the title or description.  No smoking gun yet.