nasa/XPlaneConnect #112: Pausing the physics sim for only one aircraft

MichaelRThompson, 2017-07-05:
I am looking to visualize a combination of some aircraft being controlled through MATLAB using the XPlaneConnect plugin, and some being piloted through XPlane itself.

When I try to control aircraft using the plugin, the simulation is very "jerky" if I leave the physics simulation turned on.  Meaning that the orientation is constantly shaking around.  If I turn the physics off, the plugin-controlled aircraft runs completely smooth, but obviously, I can't fly any aircraft using XPlane itself.

Have you looked into pausing the physics sim for only 1 aircraft rather than all?  I've looked through the code a little bit, but don't want to dig too deep before asking if it's something you've looked at already.

jason-watkins, 2017-07-05:
Hi Michael,

No, that's not something anyone has asked for, but I can see how it might be useful in some circumstances. I don't have enough time available for X-Plane Connect to add new features, but this one should be fairly easy to implement. If you are familiar with C++ and interested in adding this functionality, I would be glad to accept a PR.

The relevant plugin code is in [MessageHandlers::HandleSimu](https://github.com/nasa/XPlaneConnect/blob/master/xpcPlugin/MessageHandlers.cpp#L581). You can see that we currently accept `0` and `1` as direct boolean values and `2` as a toggle value. To avoid changing the size of the network message, I would probably make the values `100..119` pause the aircraft 0..19 and `200..219` unpause aircraft 0..19.

MichaelRThompson, 2017-07-05:
Jason,

Thanks for the quick response.  I'll look into this and see if I can tweak the code in the coming days.

MichaelRThompson, 2017-07-06:
It turns out that there's actually a really simple way to do this.  The DREF 'sim/operation/override/override_planepath' is an array of 20 different values, each one corresponding to an individual aircraft.  So by setting the first one to 0 and the second one to 1, you can control aircraft 1 with the plugin and still fly aircraft zero.  

The SDK wiki says to not do it this way, but as of XPlane 10, it still works.  So this project might drop a bit on my priority list.

closed 2017-07-11 by jason-watkins