nasa/XPlaneConnect #113: Errors on plugin loading and compilation

mstafford2012, 2017-07-06:
I tried to use the plugin as-is on my Ubuntu x64 machine for X-Plane 10, to no avail.  I saw the following error in the X-Plane log:

> dlerror:/home/mstafford/X-Plane 10 x64/Resources/plugins/xpcPlugin/lin.xpl: undefined symbol: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev

So I tried to recompile the plugin, and got this error:

> Linking CXX shared library xpc32.xpl
> CMakeFiles/xpc32.dir/XPCPlugin.cpp.o: In function `XPCFlightLoopCallback(float, float, int, void*)':
> XPCPlugin.cpp:(.text+0xb58): undefined reference to `__stack_chk_fail_local'
> CMakeFiles/xpc32.dir/DataManager.cpp.o: In function `XPC::DataManager::Initialize()':
> DataManager.cpp:(.text+0x3113): undefined reference to `__stack_chk_fail_local'
> CMakeFiles/xpc32.dir/DataManager.cpp.o: In function `XPC::DataManager::Get(std::string const&, float*, int)':
> DataManager.cpp:(.text+0x4a88): undefined reference to `__stack_chk_fail_local'
> CMakeFiles/xpc32.dir/DataManager.cpp.o: In function `XPC::DataManager::Set(std::string const&, float*, int)':
> DataManager.cpp:(.text+0x776f): undefined reference to `__stack_chk_fail_local'
> CMakeFiles/xpc32.dir/Log.cpp.o: In function `XPC::WriteTime(_IO_FILE*)':
> Log.cpp:(.text+0x9b): undefined reference to `__stack_chk_fail_local'
> CMakeFiles/xpc32.dir/Log.cpp.o:Log.cpp:(.text+0x401): more undefined references to `__stack_chk_fail_local' follow
> collect2: error: ld returned 1 exit status

Which I was able to fix by adding "-fno-stack-protector" to the compiling and linking options in CMakeLists.txt. Now the plugin loads correctly.  Is this a legitimate fix or just a workaround?

jason-watkins, 2017-07-06:
Hi mstafford2012,

In short: This looks like a legitimate fix.

In long: In the Linux distro we use, stack protection is disabled by default in GCC. (i.e., our compilation environment already defaults to `-fno-stack-protector`.) Many distros  (including Ubuntu, [Wikipedia gives a decent overview](https://en.wikipedia.org/wiki/Buffer_overflow_protection#GNU_Compiler_Collection_.28GCC.29)) now ship with a version of GCC that enables stack protection by default (either `-fstack-protector` or `-f-stack-protector-strong`).

Since X-Plane Connect statically links against the XPLM API libraries, it needs to match the settings that were originally used to compile those libraries. The second set of errors you got indicate that the XPLM libraries were compiled without stack protection. Explicitly specifying `-fno-stack-protector` won't hurt anything on distros that don't use stack protection by default, so there's absolutely no reason not to include it.

Feel free to submit a PR if you would like proper credit in the repo, otherwise I will add the flag myself when I have a chance.

Best,
Jason

jason-watkins, 2017-07-10:
Fixed by #117

closed 2017-07-10 by jason-watkins