nasa/XPlaneConnect #116: Multiplayer autopilot DREFs not being read or written

MichaelRThompson, 2017-07-07:
On MATLAB and XPlane 10.

Using getDREFs to read any of the multiplayer autopilot DREFs...

```
autopilot_mode | int[20] | 860-999 | yes | enum | The autopilot master mode (off=0, flight director=1, on=2)
altitude | float[20] | 860-999 | yes | ftmsl | Altitude dialed into the AP
current_altitude | float[20] | 860-999 | yes | ftmsl | Currently held altitude (remembered until you hit flchg)
vertical_velocity | float[20] | 860-999 | yes | fpm | Vertical speed to hold
airspeed | float[20] | 860-999 | yes | knots_mach | Airspeed to hold, this changes from knots to a mach number
heading_mag | float[20] | 860-999 | yes | degm | The heading to fly (magnetic, preferred) pilot
airspeed_is_mach | int[20] | 860-999 | yes | boolean | Is our airspeed a mach number
flight_director_pitch | float[20] | 860-999 | no | degrees | The recommended pitch from the Flight Director.
flight_director_roll | float[20] | 860-999 | no | degrees | The recommended roll from the Flight Director.
autopilot_state | int[20] | 860-999 | yes | flags | Various autopilot engage modes, etc.  See docs for flags
heading_roll_mode | int[20] | 860-999 | yes | enum | Bank limit - 0 = auto, 1-6 = 5-30 degrees of bank
syn_hold_deg | float[20] | 860-999 | yes | degrees | Pitch to hold in pitch-hold-mode
```
Returns a 1 x 0 empty single row vector in MATLAB.  There is also no effect when I write to them.  This occurs both before and after I write to the  `sim/operation/override/override_plane_ai_autopilot` DREF.

jason-watkins, 2017-07-07:
I'll add this to the list of things to investigate. Is there anything interesting in the XPC log? In particular, the string "ERROR: invalid DREF"?

MichaelRThompson, 2017-07-07:
A few interesting things, yes.

MATLAB Commands:

```
clear
addpath('../')
import XPlaneConnect.*
Socket = openUDP('127.0.0.1', 49009, 0, 10000);% Open connection to X-Plane
sendDREFs({'sim/operation/override/override_planepath'},...
    [0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0], Socket) 
sendDREFs({'sim/operation/override/override_plane_ai_autopilot'},[0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0], Socket)
drefs = {'sim/multiplayer/autopilot/current_altitude', 'sim/multiplayer/autopilot/airspeed'};
sendDREFs(drefs, [0 25000 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0; 0 280 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0],Socket)
getDREFs({'sim/multiplayer/autopilot/autopilot_mode'}, Socket)
getDREFs({'sim/multiplayer/autopilot/autopilot_state'}, Socket)
getDREFs({'sim/multiplayer/autopilot/altitude\'}, Socket)
```

XPC Log:

```
X-Plane Version: 10050
Plugin Manager Version: 210
Host Application ID: 1
Log file generated on 2017-07-07.
12:27:37.521| INFO|EXEC|Plugin Start
12:27:37.521|TRACE|DMAN|Initializing drefs
12:27:37.524|TRACE|SOCK|Opening socket (port:49009)
12:27:37.723|TRACE|MSGH|Setting socket
12:27:37.723| INFO|EXEC|Plugin Enabled, sockets opened
12:27:37.723| INFO|EXEC|Debug Logging Enabled (Verbosity: 6)
12:28:43.081|TRACE|MESG|Read message with length 128
12:28:43.082|TRACE|MSGH|Initializing handlers
12:28:43.082| INFO|MSGH|Handling message from 127.0.0.1:49808
12:28:43.083|DEBUG|MSGH|New connection. ID=0, Remote=127.0.0.1:49808
12:28:43.084|TRACE|DBUG| 44 52 45 46 ff 29 73 69 6d 2f 6f 70 65 72 61 74 69 6f 6e 2f 6f 76 65 72 72 69 64 65 2f 6f 76 65 72 72 69 64 65 5f 70 6c 61 6e 65 70 61 74 68 14 00 00 00 00 00 00 80 3f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
12:28:43.084|DEBUG|DBUG|Head: DREF Size: 128
12:28:43.084|DEBUG|DBUG|    DREF (size 41) = sim/operation/override/override_planepath
12:28:43.084|DEBUG|DBUG|    Values(size 20) = 4.80115e+30 2.08631e-10 4.54498e+30 6.74153e+22 2.72596e+20 2.77158e+20 1.87418e-39 0 5.83155e-39 0 0 0 0 0 0 0 0 0 0 0
12:28:43.084|TRACE|DREF|Request to set DREF value received (Conn 0)
12:28:43.084| INFO|DMAN|Setting DREF sim/operation/override/override_planepath (x:12E2EC6C) Type: 16
12:28:43.084| INFO|DMAN| -- value count was 20
12:28:43.085|DEBUG|DREF|Set 20 values for sim/operation/override/override_planepath
12:29:11.375|TRACE|MESG|Read message with length 137
12:29:11.375| INFO|MSGH|Handling message from 127.0.0.1:49808
12:29:11.376|DEBUG|MSGH|Existing connection. ID=0, Remote=127.0.0.1:49808
12:29:11.376|TRACE|DBUG| 44 52 45 46 ff 32 73 69 6d 2f 6f 70 65 72 61 74 69 6f 6e 2f 6f 76 65 72 72 69 64 65 2f 6f 76 65 72 72 69 64 65 5f 70 6c 61 6e 65 5f 61 69 5f 61 75 74 6f 70 69 6c 6f 74 14 00 00 00 00 00 00 80 3f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
12:29:11.376|DEBUG|DBUG|Head: DREF Size: 137
12:29:11.376|DEBUG|DBUG|    DREF (size 50) = sim/operation/override/override_plane_ai_autopilot
12:29:11.377|DEBUG|DBUG|    Values(size 20) = 4.80115e+30 2.08631e-10 4.54498e+30 6.74153e+22 2.72596e+20 2.57558e+20 2.85694e+32 1.81668e+25 1.23408e-26 0 1 0 0 0 0 0 0 0 0 0
12:29:11.377|TRACE|DREF|Request to set DREF value received (Conn 0)
12:29:11.377| INFO|DMAN|Setting DREF sim/operation/override/override_plane_ai_autopilot (x:12E2ECCC) Type: 16
12:29:11.377| INFO|DMAN| -- value count was 20
12:29:11.377|DEBUG|DREF|Set 20 values for sim/operation/override/override_plane_ai_autopilot
12:29:35.135|TRACE|MESG|Read message with length 245
12:29:35.135| INFO|MSGH|Handling message from 127.0.0.1:49808
12:29:35.135|DEBUG|MSGH|Existing connection. ID=0, Remote=127.0.0.1:49808
12:29:35.136|TRACE|DBUG| 44 52 45 46 ff 2a 73 69 6d 2f 6d 75 6c 74 69 70 6c 61 79 65 72 2f 61 75 74 6f 70 69 6c 6f 74 2f 63 75 72 72 65 6e 74 5f 61 6c 74 69 74 75 64 65 14 00 00 00 00 00 50 c3 46 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 22 73 69 6d 2f 6d 75 6c 74 69 70 6c 61 79 65 72 2f 61 75 74 6f 70 69 6c 6f 74 2f 61 69 72 73 70 65 65 64 14 00 00 00 00 00 00 8c 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
12:29:35.136|DEBUG|DBUG|Head: DREF Size: 245
12:29:35.137|DEBUG|DBUG|    DREF (size 42) = sim/multiplayer/autopilot/current_altitude
12:29:35.137|DEBUG|DBUG|    Values(size 20) = 7.77639e+31 1.12884e+27 3.23657e+21 7.15578e+22 2.57626e+20 7.39848e+31 1.15814e-26 0 25000 0 0 0 0 0 0 0 0 0 0 0
12:29:35.137|TRACE|DREF|Request to set DREF value received (Conn 0)
12:29:35.137|ERROR|DMAN|ERROR: invalid DREF sim/multiplayer/autopilot/current_altitude
12:29:35.137|DEBUG|DREF|Set 20 values for sim/multiplayer/autopilot/current_altitude
12:29:35.137|ERROR|DMAN|ERROR: invalid DREF sim/multiplayer/autopilot/airspeed
12:29:35.137|DEBUG|DREF|Set 20 values for sim/multiplayer/autopilot/airspeed
12:29:49.229|TRACE|MESG|Read message with length 47
12:29:49.229| INFO|MSGH|Handling message from 127.0.0.1:49808
[... cut ...]
12:30:12.873|TRACE|GETD|DATA Requested: New Request for connection 0 (1 data refs)
12:30:12.873|TRACE|DMAN|Entered Get(string, float*, int)
12:30:12.873|ERROR|DMAN|ERROR: invalid DREF sim/multiplayer/autopilot/altitude\
12:30:12.874| INFO|SOCK|Send succeeded. (remote: 127.0.0.1:49808)
12:30:18.626|TRACE|SOCK|Closing socket (1036)
12:30:18.626| INFO|EXEC|Plugin Disabled, sockets closed
12:30:18.626| INFO|EXEC|Plugin Shutdown
```

I know I made a little typo there at the end with "altitude\", but the plugin definitely isn't recognizing a lot of these DREFs as valid.

jason-watkins, 2017-07-07:
Ah, notice that according to [this list](http://www.xsquawkbox.net/xpsdk/docs/DataRefs.html), the entire `sim/multiplayer/autopilot/xxxxxx` set of DREFS are only valid in X-Plane 8 and 9.

When getting and setting DREFs, X-Plane Connect directly queries the X-Plane API to find the DREF. Any time you see "ERROR: invalid DREF" in the log, that's an indication that X-Plane itself does not recognize the DREF.

closed 2017-07-07 by jason-watkins

MichaelRThompson, 2017-07-07:
Ah, I missed that.  Wonder why they got rid of it, would have been really useful.  Thanks for the help!