nasa/XPlaneConnect #119: Aircraft continues to roll after sendPOSI

MichaelRThompson, 2017-07-11:
If an aircraft is rolling and is sent a new position with the physics sim enabled, it immediately continues to roll in the same direction as before the command was sent.

I suspect that the DREFs 
```
sim/flightmodel/position/P
sim/flightmodel/position/Q
sim/flightmodel/position/R
```
are being saved and immediately read in the following frame.  It would be a quick fix to reset these values to zero.  Is this something you think should be supported in the general use case of just sendPOSI, or should users write to P, Q, and R manually?

jason-watkins, 2017-07-11:
I'm not sure that setting the rates to 0 would be desired in all cases. I do think that it makes sense for sendPOSI to include an option for handling rates in addition to position and orientation, but I think we will have to spend some time considering how best to add that functionality.

MichaelRThompson, 2017-07-15:
Do you think that it might be more intuitive to add a new command entirely to set transitional and angular velocities?  Or alternatively, you could add a flag to the existing sendPOSI commands to allow the user to decide whether they want to reset velocities or not.

ivlkjan, 2019-10-02:
Hi, I have a similar problem with AI planes. I would like to use sendPOSI(double[] values, int ac) to initialize the position and Euler angles of an AI aircraft but when I call sendPOSI in first frame everything is set correctly however in second frame the Euler angles return to values before calling sendPOSI. Latitude and Longitude stay changed. I don't want to use override_planepaht as I only want to set the inital postion and than control the aircraft via autopilot. Is there a way how to set AI aircraft Euler angles permanently without returning back to previous values. Thank you for an answer!

jason-watkins, 2019-10-02:
@ivlkjan In the future, please open a new issue. You can set the corresponding P/Q/R drefs for AI planes:

```
sim/multiplayer/position/plane{x}_the
sim/multiplayer/position/plane{x}_phi
sim/multiplayer/position/plane{x}_psi
```

jason-watkins, 2019-10-02:
@ivlkjan In the future, please open a new issue. Unfortunately as far as I can tell there are no corresponding P/Q/R drefs for the AI planes, which means there's no way for a plugin to manually set those values.

closed 2020-06-19 by jason-watkins