nasa/XPlaneConnect #123: getDREF doesn't read in correct order

JanC, 2017-12-05:
Hi,
I'm using the C version to create a type safe wrapper in Swift. I noticed though that when requesting multiple `getDREF` in a loop, they are returned in a mixed order.

Here is a sample that reproduces the issue:
```C
void get_dref(XPCSocket sock, const char* dref, float expected) {
    
    float tVal[1];
    int tSize = 1;
    if (getDREF(sock, dref, tVal, &tSize) >= 0) {
        if(tVal[0] != expected) {
            printf("%s - expected: %.2f got: %.2f\n", dref, expected, tVal[0]);
        }
    }
}

int main()
{
	const char* IP = "192.168.1.197";    
	XPCSocket sock = openUDP(IP);

    while (true) {
        get_dref(sock, "sim/cockpit/radios/nav1_freq_hz", 10885.00);
        get_dref(sock, "sim/cockpit/radios/nav2_freq_hz", 11170.00);
        sleep(0.1);
    }
}
```

The result is randomly:
```
sim/cockpit/radios/nav1_freq_hz - expected: 10885.00 got: 11170.00
sim/cockpit/radios/nav2_freq_hz - expected: 11170.00 got: 10885.00
sim/cockpit/radios/nav1_freq_hz - expected: 10885.00 got: 11170.00
sim/cockpit/radios/nav2_freq_hz - expected: 11170.00 got: 10885.00
sim/cockpit/radios/nav1_freq_hz - expected: 10885.00 got: 11170.00
```

I could not reproduce this using the Python client so I believe this is something on the C client implementation.

I'm aware that there is the `getDREFs` method but for my project I wanted to provide a type safety around the single `getDREF` methods in Swift to facilitate the usage. The Swift API would look like this

```swift
 if let tailNum = try? connect.get(dref: "sim/aircraft/view/acf_tailnum", parser: StringParser()) {
        print("tail num: \(tailNum)") // String 'N172SP'
 }

if let nav2 = try? connect.get(dref: "sim/cockpit/radios/nav2_freq_hz", parser: FloatPraser()) {
        print("nav2: \(nav2)") // Float '11170.0'
}
```

jason-watkins, 2018-03-28:
Hi Jan,

Sorry I missed this... It sure seems like a UDP issue is to blame since the plugin just reads and responds to packets in the order it receives them. Where you ever able to come up with any additional info?

I'll try to reproduce this in the next week or two, but to be honest I don't have high hopes of being able to reproduce reliably.

JanC, 2018-03-28:
Hi,
I ended up using the `getDREFs` instead that work fine so I did not investigate further.

teubert, 2018-08-14:
I believe that the issue is the same as what we just identified with another user. If getDREF is called too quickly, it will overload the XPCPlugin. The plugin responds by clearing the requests and starting again. This can result in getting the wrong data for a request. This can be fixed by using getDREFs (as you did).

I will put a warning in the page for getDREF until we can come up with a better solution.

closed 2018-08-14 by teubert