nasa/XPlaneConnect #126: Compiling on OSX / Updated OSX Binary

klongmitre, 2018-01-18:
Very interested in leveraging XPC for some work. I'm particularly interested in controlling AI aircraft. It seems that the features added this past summer (on develop branch) enabling pausing the simulation for AI aircraft is not included in the Mac binary (mac.xpl).

For example, passing 101 pauseSim results in:
```
[12:48:54] TRACE|DBUG| 53 49 4d 55 00 65
[12:48:54] DEBUG|DBUG|Head: SIMU Size: 6 101
[12:48:54] TRACE|SIMU|Message Received (Conn 0)
[12:48:54] ERROR|SIMU|ERROR: Invalid argument: 101
```

The code looks right in MessageHandlers.cpp  on Develop. Trying to build the Xcode project (Xcode 9.3) throws an error:

```
...XPCPlugin.cpp:97:7: Call to 'abs' is ambiguous
```

Any advice/help in getting the develop branch working on Mac would be great. Thanks!

klongmitre, 2018-01-18:
Made change to XPCPlugin.cpp following advice from here: https://github.com/wesnoth/wesnoth/commit/b05773c4099e9a2c987adc0c656ec7ed96dec0ff

XPCPlugin.cpp now looks like this and the project compiles fine.
```cpp
#if (__APPLE__)
	
	int (*_abs)( int ) = & std::abs; // Needed to resolve ambiguity for OS X compilation
	
	if ( _abs(timeConvert) <= 1e-9 ) // is about 0
	{
		mach_timebase_info_data_t timeBase;
		(void)mach_timebase_info(&timeBase);
		timeConvert = (double)timeBase.numer /
		(double)timeBase.denom /
		1000000000.0;
	}
#endif
```

jason-watkins, 2018-01-18:
Hi klongmitre,

It looks like we aren't using a namespace qualifier on `abs` at the moment. Could you check whether just replacing `abs` with `std::abs` works on Mac? Seems like this probably won't work, but I would like to make sure before resorting to horrible function pointer hacks.

klongmitre, 2018-01-18:
@jason-watkins - Unfortunately, I tried that prior to the hack with no luck. Sorry.

jason-watkins, 2018-01-18:
Alright, well ugly code that compiles certainly beats code that doesn't compile at all. If you'd like to submit a pull request with the change, I'll accept it. Otherwise I'll make the fix myself later today.

klongmitre, 2018-01-18:
@jason-watkins - instead of forking for this small change, probably easier for you to just make the change. will you also commit the updated binary in the process?

jason-watkins, 2018-01-18:
Unfortunately I don't have easy access to a Mac. @teubert is the Mac guy, unfortunately he is also very busy. I'll try to remind him to get a new binary uploaded soon.

klongmitre, 2018-01-18:
Ok. No rush. I have it working for what I need for now. Others running into this issue can do the same, until an updated version is release.

commit cc797160 (2018-01-18): Resolve function ambiguity for std::abs on mac

Fixes #126
commit e1105206 (2019-08-09): Resolve function ambiguity for std::abs on mac

Fixes #126
closed 2018-01-18 by jason-watkins