nasa/XPlaneConnect #127: Run two or more X-Plane at one computer

racso10, 2018-03-27:
I use function [setCONN ](https://github.com/nasa/XPlaneConnect/blob/master/Python/src/xpc.py)by python to change one X-Plane port, and use this port to control plane.  Then I open another X-Plane, using its default port(49009),  but I cannot control it, I control the plane which I changed port! It looks like I control one plane using two port.
Why?
And what should I do if I want to control two plane in one computer?

jason-watkins, 2018-03-28:
Hi racso,

The `secCONN` function is used to change the port used on the client side. The plugin side always uses 49009. Because of this, it is not currently possible to run two instances of the X-Plane Connect plugin on the same computer.

As a workaround, you can compile the plugin twice with different values for [RECVPORT](https://github.com/nasa/XPlaneConnect/blob/master/xpcPlugin/XPCPlugin.cpp#L73).

There's no technical reason the server can't be asked to change ports, but I don't have time to implement that command myself. We do accept pull requests though.

closed 2018-03-28 by jason-watkins

racso10, 2018-03-28:
Thank you!
I do it and run three X-Plane in one computer.