nasa/XPlaneConnect #128: getDREF returning empty float array even though the dataref is not blank

mob9221, 2018-04-08:
Hello, I've recently started a project with XPC and using the Java interface for it. 

My issue is that XPC is returning empty float arrays for datarefs.

This is what I read from XPCLog.txt which seems to indicate everything is fine:

22:56:39.029|TRACE|MESG|Read message with length 17
22:56:39.029| INFO|MSGH|Handling message from 127.0.0.1:49001
22:56:39.029|DEBUG|MSGH|Existing connection. ID=0, Remote=127.0.0.1:49001
22:56:39.029|TRACE|DBUG| 47 45 54 44 ff 01 0a 70 61 6e 65 6c 73 74 61 74 65
22:56:39.029|DEBUG|DBUG|Head: GETD Size: 17
22:56:39.030|DEBUG|DBUG|    #1/1 (size:10) panelstate
22:56:39.030|TRACE|GETD|DATA Requested: New Request for connection 0 (1 data refs)
22:56:39.030|TRACE|DMAN|Entered Get(string, float*, int)
22:56:39.030| INFO|DMAN|Get DREF panelstate (x:9D33C300) Type: 32
22:56:39.030| INFO|DMAN| -- value count was 5
22:56:39.030| INFO|SOCK|Send succeeded. (remote: 127.0.0.1:49001)

But then when I go into debugging my program to see what is causing the problem, 

I see that the result array being returned by getDREF method is empty:

[image]

If someone could help me out here, It would be great :)

mob9221, 2018-04-08:
It seems to be quite random, sometimes the Datarefs are correct, sometimes they're blank, sometimes they are switched around. 

[image]

jason-watkins, 2018-04-08:
Hi mob,

Interesting that we've never seen this before, but the misordering makes me think you are having the same basic issue as #123. I have a hunch that increasing the timeout would help. Could you let me know what the result is if you run the code with a timeout of 1/2/4 seconds (note that that the `timeout` constructor parameter is in ms).

-Jason

mob9221, 2018-04-09:
Hi Jason,
I tried changing the timeout time to various seconds. The result is still the same. Sometimes gets mixed around, sometimes is blank.

jason-watkins, 2018-04-09:
Could you describe what happens with a longer timeout in a bit more detail? Is the ratio of empty results to non-empty results the same? Does the console output pause for about the timeout length between each empty response? Is it noticeably faster on non-empty responses?

I'm fairly certain that the only reasonable cause for this is that the client is timing out before receiving a response, and then sometimes it reads a previous response that is delivered late and thinks that it is a response to the current request. The nature of UDP means that this is a relatively complex problem to solve, and fixing the overall connection issue seems like a much better idea.

What is your frame-rate in X-Plane? The plugin runs in the main flight loop, so a low frame rate in X-Plane can have a negative impact on response times. You also don't show any kind of delay in your loop. Making requests in a tight loop is a sure way to over-saturate the plugin, which is essentially undefined behavior.

mob9221, 2018-04-10:
Hey Jason, 
The ratio is the same as it was before. I tried changing the time-out to 1000,2000,3000,4000 and it was the same ratio, didn't really see an affect.

My X-Plane FPS is constantly around 30. 

I have a thread.sleep in my program. I even tried making the thread.sleep 10 seconds and that didn't affect it either.

jason-watkins, 2018-04-12:
Since i can't reproduce the issue myself, I'm kind of out of idea here. Sorry I couldn't be of more help. Please let us know if you do find a way to resolve the issue.