nasa/XPlaneConnect #129: getDREFs ERROR when requesting many drefs

Clamb94, 2018-06-11:
Hi,
as soon as I request more than 102 DREFs using "getDREFs", this error occures:
"getDREFs ERROR: Response was too short. Expected at least 6 bytes, but only got 0."

Using the exact same code with 100 datarefs works fine.

jason-watkins, 2018-06-11:
The plugin reads messages into a 4096 byte buffer, and I suspect 102 drefs is enough to make the message too large to fit in that buffer. Unfortunately, looking through the code, it seems that I am not checking for that error condition.

There are two potential solutions that I can see. Either increase the buffer size and recompile the plugin, or just send two getDrefs messages. The XPC plugin runs entirely within the main X-Plane thread, so you don't want to make it do too much work in a single message. Given that, I suggest splitting up the request into a couple messages over changing the buffer size.

Clamb94, 2018-06-11:
Could you tell me where I can increase the buffer size? If possible I'd avoid splitting up the request. It just makes everything so much more complicated for my setup.

jason-watkins, 2018-06-11:
The read buffer is declared in [xpcPlugin/Message.h](https://github.com/nasa/XPlaneConnect/blob/aa3f7db02eba3583daecb11777243447286ec70e/xpcPlugin/Message.h#L47). You will need to change the bufferSize value and recompile the plugin.

Clamb94, 2018-06-11:
Didn't work for me. But I might do something wrong, that's not unlikely

jason-watkins, 2018-06-11:
Okay, next step... Could you post the XPC log file? Should be in the X-Plane plugins folder and named XPCLog.txt.

closed 2018-07-11 by jason-watkins

Svormen, 2022-04-08:
> The read buffer is declared in [xpcPlugin/Message.h](https://github.com/nasa/XPlaneConnect/blob/aa3f7db02eba3583daecb11777243447286ec70e/xpcPlugin/Message.h#L47). You will need to change the bufferSize value and recompile the plugin.

which value should I use?

jason-watkins, 2022-04-09:
There's no single answer to that question. The buffer should be big enough to hold the largest message you will send.

Svormen, 2022-04-13:
> There's no single answer to that question. The buffer should be big enough to hold the largest message you will send.
I just use getDREFs to get only few values. I get wanted values but I have one error message which is: [getDREFs] ERROR: Response was too short. Expected at least 6 bytes, but only got 0. Do you know where the problem is? Thank you so much for your answer.
My code is:
XPCSocket client = openUDP("127.0.0.1");
const char* drefs[1] = {
		"sim/cockpit2/engine/indicators/prop_speed_rpm"
	};

	float* values[1];
	unsigned char count = 1;

	values[0] = (float*)malloc(8 * sizeof(float));

	int sizes[1] = { 8 };
if (getDREFs(client, drefs, values, count, sizes) < 0)
		{
			printf("ERROR: negative return values. \n");
		}
		else
		{
			printf("Airspeed = %.f | %.f | %.f | %.f \n", values[0][0], values[0][1], values[0][2], values[0][3]);
		}