nasa/XPlaneConnect #132: Error : "initializer element is not constant"

YavuzhanERDEM, 2018-07-11:
1-#include "C:\Users\PC-359-\Desktop\XPlaneConnect-master\C\src\xplaneConnect.h" 
2-// location of xplaneConnect.h
3-XPCSocket sock = openUDP("127.0.0.1", 49007, 49008);
4-closeUDP(sock);

These are my codes to compose udp socket but when i run , error which is "initializer element is not constant" occured at 3.lines . Everything is okay for my code but I have not solved the problem.Thank you for understanding.

jason-watkins, 2018-07-11:
One issue I see is that you have the wrong combination of function name and arguments. If you want to specify ports, you need to use the `aopenUDP` function.

YavuzhanERDEM, 2018-07-13:
I have edit what you said. Unfortunately there is still a same problem.

closed 2019-07-20 by jason-watkins