nasa/XPlaneConnect #133: is it applicable to xplane11???

SASAWEN, 2018-07-18:


jason-watkins, 2018-07-18:
We have not done very much testing on X-Plane 11, but we expect it to work. Certainly the basic format of DREFs is unchanged. If you run in to any specific problems, please report them and we may be able to fix them.

SASAWEN, 2018-07-18:
Hi jason,
   I have some problem in Getting Start session...

    Installing the plugin
Binaries for the XPC plugin on all platforms are included in this repository. Installing the plugin is simply a matter of copying a few files. The plugin binaries are located in [XPlaneConnect Repository]\xpcPlugin\XPlaneConnect. Copy this entire folder to the [X-Plane Path]\Resources\plugins folder.

The plugin should now be installed. To verify, run X-Plane, and open the Plugins->Plugin Admin->Enable/Disable menu. You should see an entry named "XPlaneConnect [Version 1.X.X]"

    I have installed Xplane11 and Xplane10, but I can't find "Plugins->Plugin Admin->Enable/Disable menu" in neither Xplane11 nor Xplane10. I am wondering whether there is something wrong during my installing process?

jason-watkins, 2018-07-18:
Well, I can tell you for sure that XPC works well with X-Plane 10. What is your operating system and OS version?

SASAWEN, 2018-07-18:
Windows10  :)

jason-watkins, 2018-07-18:
Alright... next question. Could you confirm that the `X-Plane 10\Resources\plugins` folder exists (presumably so since you should have put the XPC folder there), and list the contents of that folder?

SASAWEN, 2018-07-18:
yes，already listed in here

SASAWEN, 2018-07-18:
[image]

closed 2018-07-18 by jason-watkins