nasa/XPlaneConnect #134: Sending "commands"

aholland, 2018-07-18:
Hi - just as there is a list of dataRefs ( https://www.siminnovations.com/xplane/dataref ) there is a list of commands ( https://www.siminnovations.com/xplane/command ). I think this is related to `XPLMCommandRef` in https://github.com/nasa/XPlaneConnect/blob/master/xpcPlugin/SDK/CHeaders/XPLM/XPLMUtilities.h

Is there a way to call these commands from the Java client please? I don't see anything like that in the any of the three examples. Specifically I wish to call `sim/operation/load_situation_1`.

Many thanks if you are able to help!

jason-watkins, 2018-07-18:
Hi Anthony,

There is actually an open PR for just this functionality over at #120. Unfortunately this project doesn't have active funding, so I haven't been able to squeeze in time to review and merge the PR, but perhaps you could try compiling the plugin with the PR and report back there how it goes?

If you're not comfortable working with C++, I could compile the plugin for you. I just don't have time to run any tests myself.

aholland, 2018-07-18:
Wow, that was a quick response, thanks! I am not "comfortable" compiling C++ but I would get there eventually. If it's a button press for you, please go ahead. I will gladly test it :)

jason-watkins, 2018-07-18:
[Here](https://1drv.ms/u/s!AuSTilGa7bZGm70zhryafnhdddRRrg) is a Windows/Linux build of the plugin from the PR branch. Unfortunately I don't have easy access to a Mac dev environment at the moment, but who really uses Mac anyway? :P

otisTek, 2018-07-19:
FYI I use a Mac for all kinds of things. I have way fewer problems than my colleagues that use Win10. YMMV.
I tend to do code development, old school style,  using the command line.
Also things are working great with X-Plane 11

aholland, 2018-07-19:
@jason-watkins thanks very much. Your build of the plugin and the PR branch work fine for my purposes. The commands I've tested work as expected. I too am using X-Plane 11.

closed 2018-07-20 by aholland

yvesbbbb, 2019-02-03:
@jason-watkins 
"**Here** is a Windows/Linux build of the plugin from the PR branch....."

Tip on no file !  :-(

Thank you for your code, very useful

cmgelipe, 2024-09-20:
Hi!, i am trying to send commands from java to xpc plugin, i used the code in #120 for java but I think I need to compile the plugin for mac. Could someone help me please u.u?