nasa/XPlaneConnect #135: How to conncet the x-plane from python

szsbyx, 2018-07-19:
Hello,
I have met a problem that I don't know how to connect the XPlane 10 by python language...Could you please  tell me how to solve the problem...Many thanks.
best wishes
Xiang Yin

szsbyx, 2018-07-19:
`runfile('/Users/inshou/Desktop/Python/src/basicExample.py', wdir='/Users/inshou/Desktop/Python/src')
X-Plane Connect example script
Setting up simulation
Error establishing connection to X-Plane.
Exiting...`

YoucanBaby, 2018-10-06:
Hello,
I had the same problem.Have you solved it?Could you please tell me how to solve the problem?
best wishes
Yi Fang

JanC, 2018-10-06:
- Did you set the correct IP in the `basicExample.py`? 
- Did you install the plugin in your X-Plane?
- Is X-Plane running?

This looks likes a timeout when reading the values.

adderbyte, 2018-10-09:
After a few tweaking here are the parameter settings that worked for me.

('Checking XpIp....', '127.0.0.1')
('checking Client Port ...', 1)
('checking Destination Addr ...', ('127.0.0.1', 49009))
('checking Client Addr ...', ('0.0.0.0', 1))

On the Xplane 11 go to the settings under Data output. Change the Network configurations under general data output to reflect 'client addr' above .  Also under data output ref in data output change the Network sources.
Should I have further observations I will update accordingly.

zxgcqupt, 2019-07-02:
> After a few tweaking here are the parameter settings that worked for me.
> 
> ('Checking XpIp....', '127.0.0.1')
> ('checking Client Port ...', 1)
> ('checking Destination Addr ...', ('127.0.0.1', 49009))
> ('checking Client Addr ...', ('0.0.0.0', 1))
> 
> On the Xplane 11 go to the settings under Data output. Change the Network configurations under general data output to reflect 'client addr' above . Also under data output ref in data output change the data sources to reflect client addr.
> Should I have further observations I will update accordingly.
> I hope this helps !
> check out this to see more examples : https://github.com/adderbyte/GYM_XPLANE_ML

I have followed your suggestion to change the network configurations, but the problem remains.

adderbyte, 2019-07-02:
> I have followed your suggestion to change the network configurations, but the problem remains.

 I have used XPlaneConnect (Python version - I converted it to python3 though) extensively in the project link I provided and it definitely works. May be there you are missing few important steps.
you could check how it was used under DataFilesForPretraining folder in the github page to get the intuition. Or you could outline what you did so that one could analyse where the problem might be coming from.

Thanks

zxgcqupt, 2019-07-02:
> > > After a few tweaking here are the parameter settings that worked for me.
> > > ('Checking XpIp....', '127.0.0.1')
> > > ('checking Client Port ...', 1)
> > > ('checking Destination Addr ...', ('127.0.0.1', 49009))
> > > ('checking Client Addr ...', ('0.0.0.0', 1))
> > > On the Xplane 11 go to the settings under Data output. Change the Network configurations under general data output to reflect 'client addr' above . Also under data output ref in data output change the data sources to reflect client addr.
> > > Should I have further observations I will update accordingly.
> > > I hope this helps !
> > > check out this to see more examples : https://github.com/adderbyte/GYM_XPLANE_ML
> > 
> > 
> > I have followed your suggestion to change the network configurations, but the problem remains.
> 
> I have used XPlaneConnect (Python version - I converted it to python3 though) extensively in the project link I provided and it definitely works. May be you are missing few important steps.
> you could check how it was used under DataFilesForPretraining folder in the github link provided above to get some intuition. Or you could outline what you did so that one could analyse where the problem might be coming from.
> 
> Thanks

Thanks for your reply! I downloaded GYM_XPLANE_ML and run the script keepHeading.py in the directory DataFilesForPreTraining/Sample_Scripts. Below is the output:

XPPlane Host:  127.0.0.1
checking Client Port:  1
checking Destination Addr (XPlane):  ('127.0.0.1', 49009)
checking Client Addr:  ('0.0.0.0', 1)
Press enter to end:  ...
Traceback (most recent call last):

  File "<ipython-input-1-a04481a2b021>", line 1, in <module>
    runfile('C:/Users/zhanx15/Downloads/GYM_XPLANE_ML-master/GYM_XPLANE_ML-master/DataFilesForPreTraining/Sample_Scripts/keepHeading.py', wdir='C:/Users/zhanx15/Downloads/GYM_XPLANE_ML-master/GYM_XPLANE_ML-master/DataFilesForPreTraining/Sample_Scripts')

  File "C:\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 786, in runfile
    execfile(filename, namespace)

  File "C:\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/zhanx15/Downloads/GYM_XPLANE_ML-master/GYM_XPLANE_ML-master/DataFilesForPreTraining/Sample_Scripts/keepHeading.py", line 138, in <module>
    main() # run main

  File "C:/Users/zhanx15/Downloads/GYM_XPLANE_ML-master/GYM_XPLANE_ML-master/DataFilesForPreTraining/Sample_Scripts/keepHeading.py", line 123, in main
    if sys.stdin in select.select([sys.stdin], [], [], 0)[0]:

OSError: [WinError 10038] An operation was attempted on something that is not a socket

zxgcqupt, 2019-07-02:
> I have used XPlaneConnect (Python version - I converted it to python3 though) extensively in the project link I provided and it definitely works. May be you are missing few important steps.
> you [...]

Yes, you are right. I made slight changes to the program, now it works without any problem.

zxgcqupt, 2019-07-02:
> After a few tweaking here are the parameter settings that worked for me.
>
> ('Checking XpIp....', '127.0.0.1')
> ('checking Client Port ...', 1)
> ('checking Destination Addr ...', ('127.0.0.1', 49009)) [...]

Another issue I encounter is connection error, which is caused by the p3xpc.py script. The detailed error info is listed as below. 

runfile('C:/Users/zhanx15/Downloads/GYM_XPLANE_ML-master/GYM_XPLANE_ML-master/DataFilesForPreTraining/Sample_Scripts/keepHeading.py', wdir='C:/Users/zhanx15/Downloads/GYM_XPLANE_ML-master/GYM_XPLANE_ML-master/DataFilesForPreTraining/Sample_Scripts')
XPPlane Host:  127.0.0.1
checking Client Port:  1
checking Destination Addr (XPlane):  ('127.0.0.1', 49009)
checking Client Addr:  ('0.0.0.0', 1)
Press enter to end:  ...
Traceback (most recent call last):

  File "<ipython-input-3-a04481a2b021>", line 1, in <module>
    runfile('C:/Users/zhanx15/Downloads/GYM_XPLANE_ML-master/GYM_XPLANE_ML-master/DataFilesForPreTraining/Sample_Scripts/keepHeading.py', wdir='C:/Users/zhanx15/Downloads/GYM_XPLANE_ML-master/GYM_XPLANE_ML-master/DataFilesForPreTraining/Sample_Scripts')

  File "C:\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 786, in runfile
    execfile(filename, namespace)

  File "C:\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/zhanx15/Downloads/GYM_XPLANE_ML-master/GYM_XPLANE_ML-master/DataFilesForPreTraining/Sample_Scripts/keepHeading.py", line 141, in <module>
    main() # run main

  File "C:/Users/zhanx15/Downloads/GYM_XPLANE_ML-master/GYM_XPLANE_ML-master/DataFilesForPreTraining/Sample_Scripts/keepHeading.py", line 68, in main
    tmep_action = client.getCTRL() # get the control on the simulation

  File "C:\Users\zhanx15\Downloads\GYM_XPLANE_ML-master\GYM_XPLANE_ML-master\DataFilesForPreTraining\Sample_Scripts\p3xpc.py", line 205, in getCTRL
    resultBuf = self.readUDP()

  File "C:\Users\zhanx15\Downloads\GYM_XPLANE_ML-master\GYM_XPLANE_ML-master\DataFilesForPreTraining\Sample_Scripts\p3xpc.py", line 88, in readUDP
    return self.socket.recv(16384)

ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host

Do you have any thought?

Thanks!

adderbyte, 2019-07-02:
perhaps the previous connection is still open. You need to perhaps check the open udp connections and kill the ones having port number 1 or 49009.
I hope this helps.
Regards.

zxgcqupt, 2019-07-02:
> perhaps the previous connection is still open. You need to perhaps check the open udp connections and kill the ones having port number 1 or 49009. I would suggest after running the script to check that the connection is correctly closed.
> I hope this helps.
> Regards.

I used a program CurrPorts to monitor all the ports that are in use, and found out that neither port number 49009 or 1 is in use. Please take a look at the attached figure.

[image]

Thanks!

adderbyte, 2019-07-02:
> I used a program CurrPorts to monitor all the ports that are in use, and found out that neither port number 49009 or 1 is in use. Please take a look at the attached figure.
>
> [image]
>
> Thanks!

Ok. I went through the figure. Could you filter to show only UDP connection/port ? XPlane uses port 49000 and 49001 could it be that those are the ones holding up the connection. Try killing these 2 to see if it there is any change.

Regards.

zxgcqupt, 2019-07-02:
> > > perhaps the previous connection is still open. You need to perhaps check the open udp connections and kill the ones having port number 1 or 49009. I would suggest after running the script to check that the connection is correctly closed.
> > > I hope this helps.
> > > Regards.
> > 
> > 
> > I used a program CurrPorts to monitor all the ports that are in use, and found out that neither port number 49009 or 1 is in use. Please take a look at the attached figure.
> > [image]
> > Thanks!
> 
> Ok. I went through the figure. Could you filter to show only UDP connection/port ? XPlane uses ports 49000 and 49001 for remote communication could it be that these are the ones holding up the connection. Try killing these 2 to see if it there is any change.
> 
> Regards.

OK. I will kill them now. Another question I have is whether I need to keep X-plane running prior to run the script keepHeading.py.

adderbyte, 2019-07-02:
Yes.  It is better to start Xplane before running the script.

Regards

zxgcqupt, 2019-07-02:
> Yes. It is better to start Xplane before running the script.
> 
> Regards

The two figures below show how I change the network configuration in X-Plane 11. 

[image]

[image]

Are they correct?

Thanks!

adderbyte, 2019-07-02:
This should be correct. But try also with the default configuration (without changing or adding IP in the settings): it should  normally work .

Regards

zxgcqupt, 2019-07-02:
> This should be correct. But try also with the default configuration (without changing or adding IP in the settings): it should normally work .
> 
> Regards

I followed your suggestion to kill the process, and run the script again. Unfortunately, the same problem remains. Is it caused by my firewall? 

[image]

Thanks!

adderbyte, 2019-07-02:
Ok. Perhaps it could be firewall. I guess the script already worked the first time (?). Try stopping the firewall. Restarting Xplane or the system itself. Or is any other application interfering ?

Regards

zxgcqupt, 2019-07-02:
> Ok. Perhaps it could be firewall. I guess the script already worked the first time (?). Try stopping the firewall. Restarting Xplane or the system itself. Or is any other application interfering ? Make sure you don't start 2 separate Xplane program same time, one might be blocking the other on the port .
> 
> Regards

I have attempted to restart my desktop, but the problem is still there. The script never works on my desktop. I am reading many webpages trying to figure out what caused the error. In the plugins folder, 
we only copy xpcPlugin and the folder beneath to the X-Plane 11 folder: Resources -> plugins, right?

zxgcqupt, 2019-07-02:
> Ok. Perhaps it could be firewall. I guess the script already worked the first time (?). Try stopping the firewall. Restarting Xplane or the system itself. Or is any other application interfering ? [...]

Now  the error has become "timed out".

adderbyte, 2019-07-02:
You could get instruction on how to set up Xplane for the keepHeading situation here :
https://github.com/adderbyte/GYM_XPLANE_ML/tree/master/gym_xplane_final_version

And you are right on the location of the xpcPlugin. The plugin has windows, linux and mac version of the plugin.

With the "time out" error, it is probably trying to connect and either the client or xplane is not responding. Try to disable firewall now.

Regards

zxgcqupt, 2019-07-02:
> You could get instruction on how to set up Xplane for the keepHeading situation here :
> https://github.com/adderbyte/GYM_XPLANE_ML/tree/master/gym_xplane_final_version
> 
> And you are right on the location of the xpcPlugin. The plugin folder has windows, linux and mac version of the plugin.
> 
> With the "time out" error, it is probably trying to connect and perhaps xplane is not responding. Try to disable firewall now.
> 
> Regards

Thanks for your help! We just keep X-Plane stay in the main screen, or do we need to load the scenario you created in order to make the keepHeading.py run?

adderbyte, 2019-07-02:
You could try reloading the scenario manually as a test to show the set works. 
On the Xplane menu you could select to load the scenario and also define a short cut (ctrl + k  -- or any other one ).
 If you follow the last link provided you will observe another plugin was used to reload the scenario automatically. Use this after testing with the short cut for manually reloading.

Regards

zxgcqupt, 2019-07-02:
> You could try reloading the scenario manually as a test to show the set up works.
> On the Xplane menu you could select to load the scenario and also define a short cut (ctrl + k -- or any other one ).
> If you follow the last link provided you will observe another plugin was used to reload the scenario automatically. Use this after testing with the short cut for manually reloading.
> 
> Regards

I have successfully run the keepHeading.py with your help! Thank you so much! I plan to develop a reinforcement learning-based model to correct flight trajectory. Since you already have a lot of experience along this direction, I will probably need to ask you more questions in the subsequent weeks. Thanks again!

adderbyte, 2019-07-02:
You are welcome !
I am very glad you succeeded in this first step.

Sure, you could ask questions. 
Good luck.

Regards

zxgcqupt, 2019-07-03:
> You are welcome !
> I am very glad you succeeded in this first step.
> 
> Sure, you could ask questions.
> Good luck.
> 
> Regards

Hi, 

Could you please give me your email address? I'd like to discuss a few more questions with you.

Thanks!

adderbyte, 2019-07-03:
Hi,
yes sure, here it goes : adderbyte@icloud.com
Send me  hello message after which I would delete this post.

Regards

closed 2019-07-20 by jason-watkins

praveenethiraj, 2021-01-20:
Hello!

I apologize for reopening this thread with a question again.

I am having errors conecting to X Plane with this line in the GetPOSI function definition throwing me an error:

buffer = struct.pack('4sxB', "GETP", ac)
error - argument for 's' must be a bytes function

I have tried to see what went wrong in the syntax but I find my understanding of Python to be quite lacking here. The examples that I see do have a different syntax and I am unable to actually solve the issue.

Could anyone point me in the right direction please? 

If this question is not in the right thread please do let me know. I'll try to get it in the right one.

[image]

Best regards,
Praveen

klaus-schuster, 2023-08-13:
> Hello!
>
> I apologize for reopening this thread with a question again.
>
> I am having errors conecting to X Plane with this line in the GetPOSI function definition throwing me an error:
>
> buffer = [...]

i am just having the exact same problem