nasa/XPlaneConnect #137: Running the library via Simulink

karanchawla, 2018-07-25:
Hello,

I've decided to use this lib for driving XPlane visualization via my Simulink model however I'm not sure how to send the datarefs via Simulink, do you have any experience with this? 

So the way I'm doing it right now is packing my UDP messages using the DSP toolbox and sending it to XPlane, however, it only allows for 32 bit numbers and my position is only accurate up to 0.5 m because of this. So, I'm wondering if using data refs will resolve this issue.

Thanks.

teubert, 2018-07-31:
Hello @karanchawla,

Thank you for reaching out to us!

Unfortunately, we do not have a Simulink version of the XPC. My understanding is that you can use Matlab within a Matlab Function Block in Simulink, but I haven't tried it myself. Unfortunately, I don't think this is going to solve your problem. DREFs has the same 32bit restriction.

That said, the version of XPC in the develop branch supports 64bit POSI calls. To use this you will have to build the plugin using the version of the code in the develop branch and reinstall the plugin. Then you can reform your messages to use 64bit position.

Let me know if this works for you.

rthaenert, 2018-08-04:
Hey,

invoking XPlaneConnect using a Matlab function call works fine with a persistent variable to store 
the socket connection. A blogpost on Matlab describes this in detail: https://blogs.mathworks.com/simulink/2014/01/08/simulink-and-matlab-objects/

I added an example in my local fork for a simple controller used for controlling the bank angle (https://github.com/rthaenert/XPlaneConnect).

Best Regards
Ralf

karanchawla, 2018-08-09:
Thanks a lot, both those solutions together did the job! Now I can send double lat/lon position to XPlane/HIL.

closed 2018-08-09 by karanchawla