nasa/XPlaneConnect #139: Java getDREF get same value back first 20 times, then progresses

aholland, 2018-10-31:
Hi. If I try to do the following with line 2 in a loop as XPlane is starting up, it throws exceptions before it can connect (fine) and then returns the first time it gets, 20 times, and then starts getting updated times. The updated times remain behind the time shown on screen in XPlane in the DataRefsEditor plugin.
1.  'val xpc = new XPlaneConnect()'
2.  'xpc.getDREF("sim/time/total_flight_time_sec")'

The first value is always read exactly 20 times before it starts to read in changed, new values. It doesn't matter whether I set a delay of 3 seconds between attempts, 100ms, - it gets the same value 20 times before making progress, and somehow stays behind. To me this says that there is probably a buffer of 20 UDP requests somewhere in the plugin.

Could you throw any light on this please?

jason-watkins, 2018-10-31:
Hi Anthony,

Could you elaborate on "in a loop"? The plugin runs synchronously between frames, so if you are not rate-limiting your requests on the client side it's quite possible to get 20 requests in a single frame. You are correct in a sense that there is a buffer. In fact, the plugin will process up to 20 messages from the buffer in a single frame, and then drop the rest on the floor.

aholland, 2018-10-31:
Hi and thanks. Our goal is to detect when Xplane has loaded and the default situation has loaded, so we query that data ref in a while loop with a 3 second delay until we get a vakue of more than a certain threshold (e.g. ten seconds). The first 30 or so requests result in exceptions. Then, once the situation starts, we hope to gets accurate times until one that exeecds the threshold. Instead we get a time closento zero twenty times, followed by out of date but increasing times. Is that clear?

jason-watkins, 2018-10-31:
I think I've got it, but a minimal complete example would be nice to be sure.

Unfortunately, I think the architecture of X-Plane Connect is really not amenable to this use case. I have a feeling that what you're seeing is caused by the fact that the UDP socket on the plugin side is created when the plugin is enabled, but doesn't do any processing until the flight loop starts. The trouble is, while the client may have already timed out, the plugin doesn't know that, so it sends a response anyway. This is compounded by the fact that the current wire protocol doesn't have any way of identifying which response goes with which request, so the client will pull down the response to a timed out request and return it as though it's a response to the current request. This in turn means that the next request will get the response meant for the previous request, explaining the delay once things start to update.

I don't really have a solution right now. In the long term, I'm working on rewriting the wire protocol to fix issues like this, but X-Plane Connect is just a side project for me now, so it could be many months before I have anything ready to distribute.

aholland, 2018-11-01:
Hi Jason. I have solved my problem as follows (this is XPlaneConnect.java). It would be better still to have request ids in the packets, but I didn't want to get into modifying the plugin code. The discard section below assumes that if another packet is available within ten milliseconds then the prior packet can be discarded in favour of the newer one. This happens in a loop so even if there are twenty queued up, it will discard all except the newest.

    private byte[] readUDP() throws IOException
    {
        return readUDP(false);
    }

    private byte[] readUDP(boolean discard) throws IOException
    {
        byte[] buffer = new byte[65536];
        DatagramPacket packet = new DatagramPacket(buffer, buffer.length);
        try
        {
            socket.receive(packet);
            if (discard) {
                int soTimeOut = socket.getSoTimeout();
                try {
                    System.out.println("soTimeOut=" + soTimeOut);
                    System.out.println("Setting soTimeOut to 10");
                    socket.setSoTimeout(10);
                    boolean timedOut = false;
                    int count = 1;
                    while (!timedOut) {
                        try {
                            socket.receive(packet);
                            System.out.println("Received " + ++count);
                        } catch (SocketTimeoutException ste) {
                            timedOut = true;
                        }
                    }
                } finally {
                    System.out.println("Setting soTimeOut back to " + soTimeOut);
                    socket.setSoTimeout(soTimeOut);
                }
            }
            return Arrays.copyOf(buffer, packet.getLength());
        }
        catch (SocketTimeoutException ex)
        {
            return new byte[0];
        }
    }

closed 2019-07-20 by jason-watkins