nasa/XPlaneConnect #145: socket.timeout: timed out

axita04, 2018-11-08:
Platform : Linux
Language : Python
I am repeatedly getting a timeout error while using the getDREFs() method. 
The error can be backtracked to readUDP(). I have increased the timeout in the constructor to 3000, but it still gives me the same error after a few minutes.

markaduol, 2019-02-10:
I'm having this error as well.

TheDigitalDev, 2019-03-26:
Same issue with Java version

> `    public HashMap<String, float[]> getDrefs(String[] arr) {
        HashMap<String,float[]> results = new HashMap<>();
        int chunk = 255; // chunk size to divide
        int cSize = arr.length / chunk + 1;

        String[][] chunkedDrefs = new String[cSize][chunk];

        int chunkCounter = 0;
        for(int i = 0; i < cSize; i++){
            chunkedDrefs[i] = (Arrays.copyOfRange(arr, chunkCounter, Math.min(arr.length,chunkCounter+255)));
            chunkCounter = (chunkCounter+255>arr.length) ? arr.length - (arr.length%chunk) : chunkCounter+255;
        }
        for(String[] drefs : chunkedDrefs) {
            try {
                float[][] values = xPlaneConnect.getDREFs(drefs);
                for(int i = 0; i < drefs.length; i++) {
                    results.put(drefs[i],values[i]);
                }
            } catch (IOException e){
                System.out.println("[!] Error fetching DREFs\n" + e);
            }
        }
        return results;
    }
`

Catches the IOException
java.io.IOException: No response received.

closed 2019-07-20 by jason-watkins

onionysy, 2019-09-30:
Do you fix the problem? I have the same problem with getDREFs().
Can you tell me how do you fix it?
Thank you very much! @axita04 @TheDigitalDev @jason-watkins

TheDigitalDev, 2019-10-22:
Sorry but why was this closed? Is there something we're missing that was detailed in the docs or... just any answer would be nice

jason-watkins, 2019-10-22:
In short because timeouts really have nothing to do with XPlane Connect and there's no useful advice I can give. I suppose I should add a section to the Readme or Wiki that explains that as clearly and thoroughly as possible.

axita04, 2019-10-22:
I kept increasing the timeout value in the __init__() definition until I
stopped getting the error.

YingxiaoKong, 2020-08-06:
I also got the timeout error when using client.readUDP(), anyone has any idea?

sanderdatema, 2020-08-06:
Too many requests per second will get you a timeout (if you didn't just do one request only). I build in a 'throttle' to maximize the amount to max 10/sec.

YingxiaoKong, 2020-08-06:
Thanks for your reply. I didn’t do anything and suddenly the problem disappeared. By the way, could you share with me more information about what you mean by ‘throttle’? Is this a UDP management command?