nasa/XPlaneConnect #147: UDP discovery of XPlaneConnect plugin

JanC, 2018-12-10:
Hi,
X-Plane sends a UDP multicast "beacon" packet to allow the discovery of a running X-Plane on the network. While this can be used by XPlaneConnect clients, it does not guarantee that the XPlaneConnect plugin in actually installed.

It would be a useful feature if XPlaneConnect could send a similar packet to allow the discovery of XPlaneConnect plugin.

jason-watkins, 2018-12-11:
Hi Jan,

I agree, that does seem useful. Do you have any reference document that describes what exactly X-Plane sends?

JanC, 2018-12-11:
Sure. This is the part describing the beacon format from the `Exchanging Data with X-Plane.rtfd` that comes with X-Plane

> In order to send and receive UDP messages to talk to X-Plane, you must know the IP-address of that machine within your network. You can either enter the IP-address manually, or you can use the BEACON that each instance of X-Plane running in your network broadcasts to announce its presence. 
> 
> The BEACON message uses a mechanism called multicast, which is a special IP-address range that you can subscribe to in order to get announcements from X-Plane. This works both when X-Plane is running on the same local machine, or on a machine in the same local area network.
> 
> In order to subscribe to X-Plane's BEACON, you must join the multicast group 239.255.1.1 and listen on port 49707. While this looks like an IP-address that is not in your network, it is really a group identifier. Consult your operating system's documentation on how to join a multicast group with a UDP socket. You will want to read the documentation of the setsockopt() function and the IP_ADD_MEMBERSHIP parameter.
> 
> When you configure a socket to receive X-Plane’s multicast messages, you also want to use the SO_REUSEADDR (SO_REUSEPORT on Mac) option. This is important so that multiple applications on the same machine can all receive the BEACON from X-Plane. If you don’t use SO_REUSEADDR (SO_REUSEPORT on Mac) only one application per machine will be able to detect X-Plane, and others will get a failure when they try to bind the socket. 
> However, be careful that you normally do NOT want to use any of the REUSE* options for the sockets you receive unicast traffic from X-Plane on. Unless you know exactly what you are doing, DO ONLY set SO_REUSEPORT or SO_REUSEADDR for receiving X-Plane’s BEACON on port 49707.
> 
> Once you are receiving the BEACON messages from X-Plane, the struct must be interpreted as follows:
> 
> 5-character MESSAGE PROLOGUE which indicates the type of the following struct as
> BECN\0
> 
```
struct becn_struct
{
 	uchar beacon_major_version;	// 1 at the time of X-Plane 10.40
 	uchar beacon_minor_version;	// 1 at the time of X-Plane 10.40
 	xint application_host_id;		// 1 for X-Plane, 2 for PlaneMaker
 	xint version_number;			// 104103 for X-Plane 10.41r3
 	uint role;						// 1 for master, 2 for extern visual, 3 for IOS
 	ushort port;					// port number X-Plane is listening on, 49000 by default
 	xchr	computer_name[500];		// the hostname of the computer, e.g. “Joe’s Macbook”
};
```
> 
> Parsing this struct allows you to find any instance of X-Plane running in the network, find out which version of X-Plane is running, see the name of the computer and find out whether it's configured as a master or visual slave machine, and lastly find out if X-Plane's receive port has been changed from the default of 49000.
> 
> You can expect the struct to be compatible within the same major version of the BEACON. Expect structs to change when the major version changes, so you will want to abort parsing when you discover a mismatch of the beacon_major_version.

JanC, 2018-12-15:
As a workaround, I'm thinking about implementing a client side combination of the native UDP discovery and a XPlaneConnect command to see if the plugin is installed. What XPlaneConnect command could I used that works when the user is in the main menu but fails when the plugin is not installed?

The `getDREF` commands work only when X-Plane simulation is started.

JanC, 2018-12-16:
The SDK ([docs](http://www.xsquawkbox.net/xpsdk/mediawiki/PluginThreading)) say that plugins can use threads as long as they don't call the SDK methods.

I'm not familiar with the X-Plane SDK (nor c++) but I came up with the following proof of concept that uses C++11 `thread`. 

- In the `XPluginEnable`, I start a new `Timer` that sends the `BCN` UDP to a multi cast group.

```cpp
PLUGIN_API int XPluginEnable(void) 
{
       // ...
	timer.start(chrono::milliseconds(1000), [] {
		XPC::MessageHandlers::SendBeacon();
	});
}
```

- `Timer.cpp` is just a wrapper around a thread
```cpp
void Timer::start(const chrono::milliseconds interval, const Callback &callback) {
    {
        running = true;
        th = thread([=]()
                    {
                        while (running == true) {
                            this_thread::sleep_for(interval);
                            callback();
                        }
                    });
    }
}
void Timer::stop() {
    running = false;
}
```

- Added a helper method to the `MessageHandlers`
```cpp
void MessageHandlers::SendBeacon() 
{
        // todo add host ip, plugin version ...
        unsigned char response[4] = "BCN";
        
        // todo define multicast group
        sockaddr* sa = sock->GetAddr("239.255.1.1", 49710);
        sock->SendTo(response, 4, sa);
    }
```

This compiles and runs on macOS and the packet is sent every second
```
sudo tcpdump -u port 49710 -X
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on en0, link-type EN10MB (Ethernet), capture size 262144 bytes
18:03:26.894628 IP 192.168.1.204.49009 > 239.255.1.1.49710: UDP, length 4
	0x0000:  4500 0020 1bd4 0000 0111 ea84 c0a8 01cc  E...............
	0x0010:  efff 0101 bf71 c22e 000c 3a7d 4243 4e00  .....q....:}BCN.
```

The things to do would be 
- [ ] Compile & run on Windows
- [ ] Compile & run on Linux
- [ ] Handle the `XPluginDisable`
- [ ] Define the beacon packet name and structure (ip, plugin version, ..) and fill it

opt12, 2019-05-19:
Hello everyone,
just a short question regarding the UDP multicast beacons sent out by X-Plane: Has anybody suceeded in doing this UDP discovery with X-Plane running on wondows (10) and connected to WiFi (not wired Ethernet)?

I tired this in my own software and it was no problem on Linux, but with Windows, this worked out only when I used wired Ethernet connection. So I'm just curious, whether this works with by using your approach.

(Sorry, I f I kind of hijacked this thread, but I think this kind of related in case the discovery would not work using WiFi.)

JanC, 2019-06-09:
I believe we can close this ticket as the PR has been merged.

@opt12, I had no issues running this on a wifi network (Linux and mac). I tried only a wired network on Windows but I doubt there is something in XPlaneConnect that would cause troubles on wifi. 

If you are still having issues, could you reopen a new ticket for that?

closed 2019-06-09 by JanC