nasa/XPlaneConnect #149: How to successfully position aircraft at start of runway

markaduol, 2019-01-20:
For context, I'm trying to accomplish 3 things:

1. Position aircraft somewhere on the tarmac of the airport.
2. Automatically taxi to runway and takeoff.
3. While taxiing and taking off, keep track of the distance of the front landing gear to the centre line.

For now, I'd just like some help on how to accomplish the first task.

I've looked at the examples in the repo, and I've managed to successfully position the aircraft at a location of my choosing. However, even if I position the aircraft only ~0.5m above the tarmac, once it falls to the ground, it crashes and is no longer operable. Rather than spending the next few hours fiddling with decimal points to get the correct altitude, I'd like to know whether there's a "less hacky" way to do this.

jason-watkins, 2019-01-20:
Hi Mark,

Unfortunately I'm not aware of any way to set the plane to be on the ground directly. I took a quick look at what datarefs are available and I found `sim/flightmodel/position/y_agl`, which isn't writeable, but perhaps you could set the position, read that value and then set the `sim/flightmodel/position/local_y` dref to the right value based on that.

otisTek, 2019-01-20:
Mark
I'm able to use the playback example to do what you are trying to do using tabular data. I had to turn off the physics using the pauseSim function.  In c, I placed this just before the tabular data read loop.
	pauseSim(sock, 1); // Pause simulation
Granted this only works if you want do not want to use the x-plane physics engine. This works for me as I'm computing my trajectory externally and then exploiting Jason's excellent work to make a video.

I've also had success using x-plane to position an aircraft and then use the data export function in the playback example to get the "right" altitude for my aircraft.

markaduol, 2019-01-20:
@jason-watkins You're solution worked, thanks. @otisTek You're ideas seem like they could work for me as well. Thanks, I'll try them out and report back.

jason-watkins, 2019-01-21:
Glad I could help!

closed 2019-01-21 by jason-watkins

flyingk, 2019-11-06:
A solution to this issue was just merged into develop branch.