nasa/XPlaneConnect #151: Errors on plugin loading and compilation cont.

dgriessler, 2019-02-26:
Hi, I actually took over the project from mstafford2012 who had a similar problem and posted the issue: Errors on plugin loading and compilation.

So, I've been trying to run your plugin, but the simulator wasn't detecting it.  I checked the log and this is what I found:
dlerror:/home/dgriessl/X-Plane 11/Resources/plugins/XPlaneConnect/64/lin.xpl: undefined symbol: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev
dlerror:/home/dgriessl/X-Plane 11/Resources/plugins/XPlaneConnect/lin.xpl: wrong ELF class: ELFCLASS32

I, however, don't know how to recompile it like Milton did so any ideas?

jason-watkins, 2019-02-27:
Hi dgriessler,

This error is caused by ABI incompatibility between the binary we distribute and your OS. Unfortunately there's not really a good fix besides recompiling the plugin yourself for your particular platform. Recompiling should be as easy as running the following:

```bash
XPlaneConnect$ cd xpcPlugin
XPlaneConnect/xpcPlugin$ mkdir build
XPlaneConnect/xpcPlugin$ cd build
XPlaneConnect/xpcPlugin/build$ cmake ..
XPlaneConnect/xpcPlugin/build$ cmake --build .
```

That should produce two files: `xpc32.xpl` and `xpc64.xpl`. Since you only need the plugin for your specific architecture, you should be able to delete the entire `XPlaneConnect` folder and just copy the `xpc64.xpl` file in its place.

dgriessler, 2019-02-27:
That sounds very doable.  Thank you for your swift reply.  I don't currently have cmake and am working on getting root access on this machine so I can run your prescribed build instructions.  I'll repost on here if I can't get it to work.
Just to clarify, once I am able to run those build instructions, I replace the XPlaneConnect folder in [X-Plane]/Resources/plugins with just that file xpc64.xpl?  Or, do I leave the XPlaneConnect folder in there and replace all of its contents with xpc64.xpl?

jason-watkins, 2019-02-27:
You want to remove the folder entirely and put the file in its place in the Resources/plugins directory. 

XPlane looks for plugins in two ways. If it finds a folder with the structure like the one in the XPlaneConnect folder, it tries to load the right platform-specific plugin form that folder. If you just drop a file in the plugins folder directly, it assumes that file is the right type for the platform and tries to load it. So when we're distributing the plugin to everyone, it makes sense to put all of the different plugins in a folder so that people just have to copy the folder and it (usually) "just works". Since you know which one of the plugin files you need, there's no need to bother with the folder.

dgriessler, 2019-02-27:
Ok, sounds great.  Thank you for your replies and I will attempt them just as soon as I have root access.

dgriessler, 2019-03-28:
I was finally able to test those instructions and they do work.  Thank you very much for your help.

closed 2019-03-28 by dgriessler