nasa/XPlaneConnect #152: Is it posible to get Longitude in double?

huaiyuat, 2019-02-27:
When doing 
getDREF() in matlab for ' sim/flightmodel/position/longitude'.

The result is returned in single/float32. (10 meter precision on earth)
However XPSDK wiki states it is a double/float64.

Is there a way to get the double value?

jason-watkins, 2019-02-27:
We made the decision originally to convert everything to single-precision because that was all that was needed at the time. The `develop` branch has been changed to upgrade everything to double precision. If you build the plugin from that branch and use the client there you should get a double precision number.

closed 2019-02-27 by jason-watkins

huaiyuat, 2019-02-27:
I downloaded developer branch, replaced the old plugin and did get double
from getPOSI function. But it is still with precision .0001 . Did I missed
something here?

jason-watkins, 2019-02-27:
>it is still with precision .0001

Precision where exactly? whether single or double precision, the value is certainly grater precision than that. Is it possible that you're looking at a string representation that has been printed at reduced precision?

Also, for reference what is the approximate longitude that you're trying to retrieve?

huaiyuat, 2019-02-27:
My bad. Matlab hides the rest digits automatically.

Thank you very much

dfremont, 2019-09-25:
> The `develop` branch has been changed to upgrade everything to double precision. If you build the plugin from that branch and use the client there you should get a double precision number.

We also need latitude/longitude in double precision, but it looks like the `develop` branch still doesn't provide this: the values are cast to `float`s [here](https://github.com/nasa/XPlaneConnect/blob/c9020e6ca7023a879e2b73ba394824ef3d0e0aad/xpcPlugin/MessageHandlers.cpp#L560). Do you plan to fix this in the near future, or possibly provide a general function for obtaining any dataref in its full precision? Thanks!