nasa/XPlaneConnect #154: Question: sending and overwriting way points

dgriessler, 2019-03-11:
Hi,

You feature the ability to add and clear the way points in X Plane.  In X Plane, I have a loaded flight plan, and what I need to be able to do is change the flight plan by using this command to change the flight plan way points to a new set of way points.

So, my question is: How can I accomplish the overwriting of the flight plan?  Do I use the sendWYPT command?  If so, this is my current attempt and it doesn't do anything noticable to the current plan:
xpc.sendWYPT(WaypointOp.Clr, new float[]{});
xpc.sendWYPT(WaypointOp.Add, newPoints);
Where newPoints is a declared array as you are expecting triplets of latitude, longitude, and altitude in that order.
Now, if it's the flight plan, then I don't need an initial flight plan if I can also use the sendWYPT command at the beginning of my program to send a set of way points that act as the flight plan.

The only way I've found that makes a difference is that if I change the .fms file that is loaded in X Plane and then manually reloading the flight plan in X Plane.  However, this isn't ideal because the command to overwrite the way points are being done by an autonomous agent that is supposed to continue taking courses of action after overwriting the way points.  If I had to manually reload the flight plan, then it weakens the automation. 

A sub question is: What way points are you supposed to send when you send the opcode WaypointOP.Clr?  I just assumed to send an empty array since the first thing that the method checks for is that the size is divisible by 3 or throws an error, but I wasn't sure.

Sincerely,
Daniel Griessler

jason-watkins, 2019-03-11:
Hi Daniel,

Perhaps the terminology isn't as clear as it should be. The "waypoints" that XPC deals with are only visual markers that are drawn in the world. They don't affect the plane's flight plan at all.

dgriessler, 2019-03-11:
Hey Jason,
Ok, my mistake.  Is there anything that exists that could do what I'm trying to do?

closed 2019-07-20 by jason-watkins