nasa/XPlaneConnect #158: limited rate in monitor example

alireza787b, 2019-04-30:
Hi
It sems the receive packet rate wont change even with changing the pause value... its about 2-4 Hz... is there a problem with XPC plugin or xplane itself?

jason-watkins, 2019-04-30:
Hi,

Hard to say without more detail. What is the "pause value" you are referring to? The XPC plugin will only process a fixed number of messages per frame, but for any reasonable framerate that should translate to a great deal more than 4 messages per second.

alireza787b, 2019-04-30:
I am using following code:
Socket = openUDP()

figure('units','normalized','outerposition',[0 0 1 1]);

DREFS = {'sim/time/total_running_time_sec',...    
         'sim/cockpit2/gauges/indicators/airspeed_kts_pilot',...
         'sim/cockpit2/gauges/indicators/altitude_ft_pilot',...
         'sim/cockpit2/gauges/indicators/vvi_fpm_pilot',...
         'sim/weather/temperature_ambient_c',...
         'sim/cockpit2/engine/indicators/torque_n_mtr',...
         };

while 1
   
    try 
   % posi = getPOSI(0, Socket);
   % ctrl = getCTRL(0, Socket);
    result = cell2mat(getDREFs(DREFS));
    catch
        continue;
    end 
....

pause(0.05)
end

the simulation frame rate is about 20-25. but data output in Matlab is about 5-10 Hz?
does UDP communication has a limitation frequency?

closed 2019-07-20 by jason-watkins