nasa/XPlaneConnect #160: Setup build CI

JanC, 2019-05-17:
Hi,
I open this ticket so that we can continue the conversation started in https://github.com/nasa/XPlaneConnect/pull/138#issuecomment-492814791
(cc @tukan68)

My initial Travis setup compiles now fine on macOS and Linux. 
https://travis-ci.org/JanC/XPlaneConnect/builds/533716547

I started to look into Windows but it looks like Travis has a limited support for building Visual Studio solution. I'll need to investigate more.

Once all three platform build fine, we can discuss on which basis  the builds are triggered and when/how we distribute the binaries. A proposal would be 
- Each PR triggers a build
- Each merge to develop or master triggers a build + [Github release](https://docs.travis-ci.com/user/deployment/releases/)

jason-watkins, 2019-05-17:
This is a great start! Another thing on my todo list is to remove the hand-maintained XCode/Visual Studio projects and put everything in CMake, so if that would make getting the Windows build working on Travis easier feel free to do so.

I think we want to have each PR trigger a build just to make sure that the plugin does build. Eventually maybe we can even get some tests written that don't require X-Plane to be running.

I also like the release idea. I would definitely like to have a release created for every merge to master. Perhaps we could also do "beta" releases for each PR merged to develop, but I'll have to think about that a bit more.

JanC, 2019-05-18:
I was also thinking that one CMake for both linux and Windows would be a cleaner approach. It appears though that the Travis build using Visual Studio's `MSBuild.exe` actually works. So the common CMake can remain as an enhancement.

So now we have CI builds for all three platforms 🎉 

https://travis-ci.org/JanC/XPlaneConnect/builds/534123811

Next step is to gather all binaries from the different jobs, zip & publish them. To my surprise, Travis does not seem to support [sharing files between jobs](https://docs.travis-ci.com/user/build-stages/share-files-s3/) locally so each job must upload its output file to aws s3 bucket then download them all again.

tukan68, 2019-05-19:
I thought about investing the weekend, but Jan has done almost everything. Many thanks.
Too bad that Travis can not simply summarize artifacts from different builds. The reason is probably that the builds are built on different servers and it is easier to merge the files via a cloud service.

JanC, 2019-05-19:
So the bad news is that the Windows builds somehow stoped working even if it was working quite fine. yesterday. I asked on the [Travis forums](https://travis-ci.community/t/all-windows-commands-stopped-working/3448). 

The good news is that I managed to automate a Github release that includes the compiled binaries. Basically each build step uploads its binaries to a aws bucket. The last deploy step downloads them again, zip them into a `XPlaneConnect.zip` and the creates a Github release with that file. 

Here is an example on my fork. It includes only macOS and Linux binaries but you get the idea
https://github.com/JanC/XPlaneConnect/releases/tag/v1.3.0

The Travis build looks like this
https://travis-ci.org/JanC/XPlaneConnect/jobs/534532060

JanC, 2019-05-21:
It seems that Windows on Travis is in a experimental stage and I was not able to get a stable Windows pipeline. It even stopped working completely.

appveyor.com on the other hand works quite fine and fast for windows but they do not support macOS. 

I came up with a solution where a script on Travis would trigger a build on AppVeyor and then get the compiled binaries.  Basically:
- Travis step: build macOS
- Travis step: build Linux
- Travis step: trigger AppVeyor build with commit hash + download the binaries
- Travis deploy step: gather all binaries

Here is a Travis build with the [three build steps](https://travis-ci.org/JanC/XPlaneConnect/builds/535495344). The 3rd python step trigered [this build](https://ci.appveyor.com/project/JanC/xplaneconnect/builds/24711559) on AppVeyor 

I still have to finish the python script that downloads the binaries from AppVeyor but I think I'm getting closer to a stable solution.

It might seem a bit hacky but at least it works.

JanC, 2019-05-23:
I have a working solution described above. 

This is an example of a CI build triggered by a pull request
https://github.com/JanC/XPlaneConnect/pull/6 
https://travis-ci.org/JanC/XPlaneConnect/builds/536285822

And this is a test release made by the CI when a tag is pushed. The zip contains all binaries
https://github.com/JanC/XPlaneConnect/releases/tag/v1.3.0
https://travis-ci.org/JanC/XPlaneConnect/builds/536283170

JanC, 2019-05-24:
I've just tried MS Azure and it turns out it works quite well on all three platforms without any workarounds or S3 accounts. 
https://dev.azure.com/JanCC/XPlaneConnect/_build/results?buildId=35&view=logs

I might go for that instead

JanC, 2019-05-31:
The MS Azure seems to be a better solution so I created a PR. 

@jason-watkins any chance that you review it and set up the CI on in the repository?

jason-watkins, 2019-05-31:
It's on my todo list to look at it this weekend.

JanC, 2019-06-13:
Did you manage to setup the MZ Azure [pipeline](https://github.com/nasa/XPlaneConnect/pull/162#issuecomment-498016386)? I see you pushed some commits in a open PR which should have triggered a pipeline build

jason-watkins, 2019-06-14:
Working on it... It seems I'm missing some permissions at the organization level to make Azure happy. Still trying to find out if I can get those. If not, my fallback plan is to figure out some kind of solution where I can pull PR's into my personal fork and do builds from there.

jason-watkins, 2019-06-14:
For reference in case I'm missing something obvious, this is the error message I see when trying to run the pipeline:

[image]

mrusme, 2019-06-30:
@jason-watkins do you have admin access to this repository? Azure will most likely need this in order to configure the hooks it requires to get notified about changes by GitHub.

JanC, 2019-07-15:
@jason-watkins did you try to troubleshoot this error further?

jason-watkins, 2019-07-15:
Thanks for poking me. I've been annoyingly busy at work so I haven't made much progress. I did manage to find out who has the ability to grant me additional permissions on the repo. I just sent them an email. Hopefully I'll hear back soon.

JanC, 2019-07-15:
finger crossed 🤞 It would be a pity to drop this CI support so close to a working solution. Thanks for your feedback

jason-watkins, 2019-07-20:
Still no definitive answer on the NASA side. In an effort to not let the perfect be the enemy of the good, I've gotten the Pipeline working on my personal fork. This means there's still a bit of manual work on my part to create the release over here, but at least I can now get the complete plugin archive with a single click.

There's now a new 1.3 RC2 release published [here](https://github.com/nasa/XPlaneConnect/releases/tag/v1.3-rc2). I'm trying something new here in that this release doesn't come with the level of testing that previous releases have had. I believe the code review process has kept the code quality high, but it's possible some bugs have slipped through. I'm hoping that those of you with a bit more time to devote to this project than I have will give it a spin and report any issues.

Thank you very much @JanC for getting the CI working. Without it I just wouldn't have had time to get this release out the door.

jason-watkins, 2019-07-20:
I'm going to go ahead and close this since the CI is working. If I can get the necessary approvals, it should be trivial to shift it over to use the NASA repo directly.

closed 2019-07-20 by jason-watkins

JanC, 2019-07-21:
awesome, thanks for taking the time for this