nasa/XPlaneConnect #166: Linker error with Visual Studio 2017

yvesbbbb, 2019-08-03:
Hi Jason,
The compilation is Ok  but the linkage display this error !?

error LNK2001: unresolved external symbol "public: void __cdecl XPC::Timer::stop(void)" (?error LNK2001: unresolved external symbol "public: void __cdecl XPC::Timer::start
....
1>..\XPlaneConnect\64\win.xpl : fatal error LNK1120: 2 unresolved externals
1>Done building project "xpcPlugin.vcxproj" -- FAILED.

Have a nice day,

Thank you

jason-watkins, 2019-08-03:
Thanks for the report. I’m out of town this weekend so i can’t verify this right now. @JanC, am I remembering right that you added the Timer as part of the Beacon code? Do you have any ideas?

yvesbbbb, 2019-08-04:
Take your time, it's not a rush.
Thank you

JanC, 2019-08-04:
Hi,
yes the `Timer` class was added recently. I’ll try to reproduce the error and will report back.

JanC, 2019-08-04:
@yvesbbbb could you please give me a link where I can download MS Visual Studio 2017? All the links I found on https://visualstudio.microsoft.com/ point to Visual Studio 2019.

Also which configuration (`Debug` or `Release`)  and which platform (`Win32` or `x64`) do you build when you get this error?

yvesbbbb, 2019-08-04:
Hi JanC,

I used the release X64 configuration with Windows 10 and the lastest update of VS2017

Thank you,

Yves

JanC, 2019-08-05:
So we have the same setup but your fails. Could you please post en entire build log?

yvesbbbb, 2019-08-05:
Hi JanC,

The complete log is the attach file.

Thank you for your time !

[Yves]
[LinkErrors.txt](https://github.com/nasa/XPlaneConnect/files/3468911/LinkErrors.txt)

JanC, 2019-08-05:
Could you double check that
- you have a unmodified version of the repo on the `master` branch
- the `Timer.cpp` file is correctly in the ` Source Files` tree in Visual Studio?

[image]

yvesbbbb, 2019-08-05:
Exactly the same ;-)
[image]

JanC, 2019-08-05:
```
1>x64\Release\Timer.obj : warning LNK4042: object specified more than once; extras ignored
```
I also see this line in the build log but it's suspicious. I noticed that the Item Type of `Timer.h` is set to `C/C++ compiler` while it should be `C/C++ header`.

Could you right click on `Timer.h` in VS,  select `Properties > General`  and change the `Item Type` to `C/C++ header`. Then clean and build again.

[image]

yvesbbbb, 2019-08-05:
Bingo !!!
You got it !!

1>Generating code
1>Finished generating code
1>xpcPlugin.vcxproj -> E:\Mes documents\FS\Xplane\GIT\XPlaneConnect-master\xpcPlugin\xpcPlugin\..\XPlaneConnect\64\win.xpl
1>Done building project "xpcPlugin.vcxproj".
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========

Many thanks !!   ;-) ;-)

JanC, 2019-08-05:
de rien ;)

Great! I created a pull request to fix this.

jason-watkins, 2019-08-05:
Thanks to both of you for getting this figured out. I've merged the fix into develop. I'll push out a new RC tonight to update master.

The mysteries of Visual Studio are a long standing frustration to us all. I'm not happy about it, but I've basically given up trying to understand *why* Visual Studio does anything.

PR #167 (2019-08-05): Fixed Timer.h Visual Studio Item Type
A wrong `Timer.h` file type in Visual Studio was causing linker errors.

fix #166
closed 2019-08-05 by jason-watkins