nasa/XPlaneConnect #173: Connecting to X-plane 11 from Matlab

ishit07, 2019-08-14:
Hi, I am new to X-plane and Matlab as well.
I am trying to connect to x-plane 11, I have connected my PC(PC1) via Lan to the PC(PC2) running Xplane. 
I have copied the plugin as per describes in the instruction. I am able to see the plugin from the simulator under Plugins--> PluginAdmin--> Enable/Disable.
I have given my PC1 ip address to DataOutput window as well under the Settings. 
Now I am trying to connect xplane-11 using Matlab.
But the I am trying this line of code --> Socket = openUDP() 
Its giving me error in Matlab.
[image]

jason-watkins, 2019-08-14:
Aside from repeating the error message back to you, I'm not sure what help I can provide. XPlaneConnect certainly doesn't define multiple "openUDP" functions.

ishit07, 2019-08-14:
Thanks @jason-watkins for a quick reply.
Can you let me know if I am going in the right direction?
So we just need to install the plugin in the pc where we have xplane-11 installed.
Then on another pc where we want to read data can we just run this line Socket= openUDP() or we need to install some library before using this function.

jason-watkins, 2019-08-14:
As far as I can tell, yes you are headed in the right direction. It sounds like the plugin is correctly installed. If you see it in the list, it's installed; not much else to it there. On the client machine, the only dependency is the entire +XPlaneConnect folder from the MATLAB section of the repository.

Have you tried running the Matlab examples? Do they give the same error?

ishit07, 2019-08-14:
Yes, it's giving me the same error.

[image]

jason-watkins, 2019-08-14:
I've done about as much investigation as I can without being able to reproduce the issue and I still don't have any real answer for you. It seems that the error message usually happens when the function name doesn't match the file name or the file has extra stuff besides the function in it, but neither of those things are the case here.

The only other possibility seems to be that there is already a function called "openUDP" defined somewhere, but it's not in XPlaneConnect and it's not a builtin function so there's really nothing I can do to troubleshoot why that would be.

ishit07, 2019-08-15:
Thanks, @jason-watkins, I really appreciate your efforts. I will try to rectify it and if I get the thing then I will update it here.

Andrew738, 2019-08-16:
Hi, i am new to XPC as well and have a sample running. I am just began learning c++ ;-)

Maybe you can try to look at the declaration of OpenUDP on line 13. In line 7, xplaneConnect.* is included which also having xplaneConnect.h. In this header file, openUDP is already declared...
What i know from C++ is that declaration is the full definition so "openUDP(const char *xpIP)" which ends up in "openUDP('127.0.0.1', 49009)' and therefore duplicated

Just an idea...

closed 2019-09-30 by jason-watkins