nasa/XPlaneConnect #174: Error LNK2019	unresolved external symbol _openUDP referenced in function _main

Andrew738, 2019-08-16:
Hi,

When i try to build the following i get a error "LNK2019	unresolved external symbol _openUDP referenced in function _main". I have included the xplaneConnect.h file

#include <iostream>
#include "xplaneConnect.h"

int main()
{
	XPCSocket sock = openUDP("127.0.0.1");

	// Get the status of the gear handle
	const char* dref = "sim/cockpit/switches/gear_handle_status";
	float value = 0.0F;
	int size = 1;
	if (getDREF(sock, dref, &value, &size) < 0)
	{
		printf("And error occured.");
	}
	else
	{
		printf("The gear handle status is %f.", value);
	}

	closeUDP(sock);
}

jason-watkins, 2019-08-16:
Hi Andrew,

How are you compiling the code? It sounds like you are only compiling the main file. You need to compile both your main file and the xplaneConnect.c file and link both of them. The exact way that you do this depends on what build tools you're using.

jason-watkins, 2019-08-16:
Also, please open a new issue for any questions you have (as you've done here) rather than commenting on an unrelated issue. I've deleted your other comment on #173, but feel free to re-post it as its own issue.

Andrew738, 2019-08-19:
Hi Jason, 

Will do. 

Thanks, i think i forgot the xplaneConnect.c file. Just new to visual studio 2019. Programmed always with xcode ;-)

Andrew738, 2019-08-19:
Works and you can close the issue ;-)

jason-watkins, 2019-08-19:
Glad to hear you were able to figure it out.

closed 2019-08-19 by jason-watkins