nasa/XPlaneConnect #177: openUDP Matlab

BDUBS9182, 2019-09-25:
I am having a problem that whenever I try running the MonitorExample.m in MatLab, I get an error in the getPOSI function where it says "no response received".

jason-watkins, 2019-09-30:
Have you verified the plugin is running? Have you looked at the plugin log?

BDUBS9182, 2019-10-02:
The plugin is not showing up in X-Plane even after i have installed it into the plugin folder. I restarted X-Plane, went to my flight, hit the plugin tab, and do not see anything.

jason-watkins, 2019-10-02:
Okay, what OS/CPU architecture are you using? Could you post the X-Plane log file?

BDUBS9182, 2019-10-09:
I am using Windows 10. Attached is the Log

[Log.txt](https://github.com/nasa/XPlaneConnect/files/3707523/Log.txt)

closed 2020-06-19 by jason-watkins

vibes99, 2021-07-21:
Hi, I am also facing very similar issues in connecting from Matlab to X-plane. I've tried running MonitorExample.m in my MATLAB R2021a and got "no response received". I've copied the XPC plugin into the plugins folder but it does not appear in the Plugin Admin tab of X-plane 11. It might be something silly but I am new to both software and any help would be appreciated.