nasa/XPlaneConnect #178: about reset theflight

onionysy, 2019-09-29:
Hello, thanks for your open code.
I tried to reset the flight with the python command. But I have try in some methods but still not work, can you give me some advice.

onionysy, 2019-09-29:
I mean when the plane is not crash.

jason-watkins, 2019-09-30:
There's really not enough information here for me to offer any advice.

closed 2019-09-30 by jason-watkins

onionysy, 2019-09-30:
> There's really not enough information here for me to offer any advice.

I want to implement such a feature in python.While the plane is still flying, reset the plane to its original departure point and start a new flight.I have tried to reset the position and speed of the plane directly before, but there will still be a small initial speed when reset to the original point.