nasa/XPlaneConnect #180: Plane flickering in playback of a recorded flight

PauloLemgruberJeunon, 2019-10-04:
Hello,

Sorry to bother, but I am having a problem with record/playback functions of the xpc API. When I record a flight with getPOSI and then try to play it back on the simulator without the physics engine running, the plane starts to flick in the horizontal plane by some delta less than 1 meter while moving. This flicks are small teleportations in random locations of the horizontal plane with a maximum distance of one meter of the center of the aircraft, but the mean position stays the same. This issue does not happen when the airplane continuously
[image]
[image]

 receives a constant position.

I think this is due to floating values having poor precision for this kind of situations where the horizontal position of the aircraft is being set by latitude and longitude in degrees. 

Is there another way to receive and set the position of the aircraft? I imagine this can be solved by either receiving/sending the data in double format or by receiving/sending the position data in local Cartesian coordinates, but I could not find a way of doing this with the provided API.

Sorry again to bother with this issue and thanks a lot for this amazing API!

PS: If there is a way to do what I intend but it is not yet implemented, I would be happy to contribute to add this feature.

jason-watkins, 2019-10-04:
Hi Paulo,

Are you using the latest release? If so, the position lat/lon/alt should be already be sent as doubles. Adding a new message type to allow setting the local position directly is certainly possible and might aid with verifying that double precision still isn't enough.

If you're interested in doing so, I'd suggest starting by reading [this](https://github.com/nasa/XPlaneConnect/wiki/Network-Information) page on the wiki that explains how the messages are formatted. Once you understand the message format, It should be relatively easy to add a new message to the [MessageHandlers](https://github.com/nasa/XPlaneConnect/blob/master/xpcPlugin/MessageHandlers.h) class. Finally you would need to add the ability to send that message type to whichever client you are using.

PauloLemgruberJeunon, 2019-10-07:
Hi Jason, 
Sorry for the delay. Yes I am using it the **1.3 RC5** release. Thanks a lot for the insight!

The C API is updated and sending the data as doubles as you have sad, but the Python API (that is the one I am using) is not. So I made a small modification and now it is working fine. The discretization delta of double precision values is small enough to eliminate the flickering.

The other issue that I am having now is to get the position with double precision values (now they are  being sent as 4 byte floats). But I am looking at the xpcPlugin and it appears to be easy to change it to enable the use of double precision on the **MessageHandlers::HandleGetP** method or to create an entire new method.

Thank you a lot for your help!

pencey, 2019-11-07:
Paulo could you please provide the modifications you did to the python code? I am also getting stuttering.

PauloLemgruberJeunon, 2019-11-07:
Hello, @pencey, yes, of course! I had to modify the plug-in to be able to receive the data in double format, so you will have to compile it to generate the modified plug-in too. I created a fork of this repo to make the modifications, follow this link: https://github.com/PauloLemgruberJeunon/XPlaneConnect

It is on a branch called "get_double_posi".
Good luck and if you have any trouble, please, contact me :) .

pencey, 2019-11-08:
That's great! Thank you!!

jason-watkins, 2020-07-15:
I think this has been resolved by #202. Feel free to comment/reopen if there's still something that needs to be resolved here.

closed 2020-07-15 by jason-watkins