nasa/XPlaneConnect #181: Difference between latitude & longitude sent by sendPOSI and read by getPOSI

yuhao-z, 2019-10-14:
Hello,

I'm currently using X-plane Connect with X-Plane 11 in MATLAB (R2018b) and there is an issue I found when I used the 'sendPOSI' and 'getPOSI' commands. 

First, I used 'sendPOSI' to place the regarding aircraft at a specific longitude & latitude while X-Plane was paused. Then without moving the aircraft and keeping X-Plane paused, I used 'getPOSI' to read the current longitude & latitude. However, those two longitude & latitude pairs are not exactly the same.

For example, using following commands:
_posi_send = [42.2257528, -83.7499732, -998, -998, -998, -998, -998];
sendPOSI(posi_send, 0, socket);
posi = getPOSI(0, socket);_

'posi' will return [42.2257538, -83.7499695, ~,~,~,~,~]; (The elements represented by '~' are not concerned.)

As you can see, longitude & latitude in 'posi' and longitude & latitude in 'posi_send' are not the same. Is this caused by some round-off error or I'm not using X-Plane Connect correctly?

Really appreciate it if you could look into this issue at your convenience.

Thanks,
Yuhao

jason-watkins, 2019-10-14:
That does in fact look like floating point rounding error. 32 bit floats have ~7 digits of decimal precision, so you would expect the 8th digit to be off. The C client supports sending/getting position with 64 bit precision now. I'd gladly merge a PR that updated the Java and MATLAB clients to using 64 bits, but it's not something I plan to do myself.

yuhao-z, 2019-10-15:
Thanks for your reply.

However, when I tried the C client (Version 1.3-rc.1),  I still couldn't get position with 64 bit precision. (I also see a comment in xplaneConnect.c: '// TODO: change this to the 64-bit lat/lon/h' in function getPOSI).

PauloLemgruberJeunon, 2019-11-10:
Hello, @yuhao-z, sorry to bother you, but I made a fix (change from double to float) for this in a fork in my github. I commented about it on issue  #180 and I made a pull request with this modifications on #189.

I hope that my changes are relevant to solve your problem.
Have a nice day :)

closed 2020-06-19 by jason-watkins